SBFspot/SBFspot #2: UploadDaemon not uploading some records with pvoutput=null

hummela, 2017-10-03:
Hi,

I have data missing in pvoutput.org. To be sure I first queried the converter to (re)get the data with the SBFspot -ad65 option (in my case 65 days would more than cover the missing data). I also tried deleting the last few months in the database and reget the data from the converter. The data is visible in the database and has the pvoutput field set to null. However, the upload daemon won't upload this data. Is this because after the set of 'null' values there are more recent records with pvoutput=1 and the 'gap' of null values is not seen? I would expect the upload daemon to upload everything that has pvoutput value null. 

To be clear, the daemon is uploading recent records just fine.

https://www.pvoutput.org/list.jsp?userid=45550
Data missing from 25 aug to 20 sep this year. This data is available in the sqlite database (with the null values).

Also am I right in assuming that the daemon will just upload everything without 'negotiating' with pvoutput.org if the records are needed?

Regards,
Arjan

sbf-, 2017-10-03:
PVoutput has a limit of 14 days upload, unless you donated.
See [help](https://pvoutput.org/help.html#donations-api-bonus)
> Live Backload
The number of days into the past live data may be uploaded is increased to 90 days (from 14).

> Also am I right in assuming that the daemon will just upload everything without 'negotiating' with pvoutput.org if the records are needed?

Yes, this will be a feature of V4

hummela, 2017-10-03:
PVoutput.org gladly took my money, and the effect is shown in my account page (donation = active). I stopped the upload daemon and changed all the pvoutput field values to null (from the missing date to today). However, there is no effect. Still only the last 14 days are uploaded. I also notice that it still uploads 30 datapoints at a time (instead of the expected 100).

I will wait a while since this could also be a pvoutput.org/API issue. Maybe my donation is not (yet) reflected everywhere where it should be. Will update here soon.

sbf-, 2017-10-03:
**`SELECT * FROM Config;`**
will show you Batch_DateLimit and Batch_StatusLimit (should be 90 and 100 in your case)
No need to restart the daemon (NextStatusCheck) but it won't hurt of course.

closed 2017-10-04 by sbf-