SBFspot/SBFspot #3: Three string SMA inverters

jrmole, 2017-10-14:
Hi,

I just had a new SMA inverter installed yesterday that replaced three older units (the 17 year-old one had finally died).  I wanted better logging than the new inverter supports and came across this software.  Thanks so much, but it only grabs the data from the first two strings.

I've made the following changes, but seem to be missing something.  And I should add that the last time I worked in C or C++ regularly was 1994, so I am a little rusty.

In SBFspot.cpp

Around line 425:
   Was: "Inverters[inv]->calPdcTot = Inverters[inv]->Pdc1 + Inverters[inv]->Pdc2"
   Is:      "Inverters[inv]->calPdcTot = Inverters[inv]->Pdc1 + Inverters[inv]->Pdc2 + Inverters[inv]->Pdc3;"
   Added extra printf line for the String 3 information (copied from String 2 and modified)

Around line 2600:
  In "SpotDCPower" case
     Was: "last = 0x00251EFF;"
     Is:      "last = 0x004524FE;"
  In "SpotDCVoltage" case
     Was: "last = 0x004521FF;"
     Is:      "last = 0x004524FE;"
  I was just guessing that the data structures would be the same size and that seemed to work...

Around line 2900:
  In "DcMsWatt" case:
     Added a case for cls == 3
  In "DcMsVol" case:
     Added a case for cls == 3
  In "DcMsAmp" case:
     Added a case for cls == 3

I was just starting to work on the CSV and SQL sections, but feel like I am not following the flow of the code properly.

So, is there an easier way to go about this update?  I am glad to dig in or open up the firewall so you can address my inverter directly to help with the debugging the code against the new model.

Thanks

Randy

sbf-, 2017-10-16:
This is a rather big change as it requires a schema update to store the extra data.

scotte, 2019-06-27:
I'd like to see this feature as well.

scotte, 2019-08-01:
I'm working on this and will submit a pull request once it's all working. As noted above, there are a lot of bits to change, so it'll take me some time to get through everything...

sbf-, 2020-07-20:
See #401 for a more versatile solution

closed 2020-07-20 by sbf-