SBFspot/SBFspot #4: SBF Spot Stopped uploading

fritserasmus, 2017-10-22:
My Raspberry Pi stopped uploading data collected from my two SMA inverters. It stopped after updating ONE reading this morning: 2017/10/11
The log updates with error messages. 
Herewith the first entries of the day:
_**[05:45:38] INFO: Uploading 2 datapoints, starting with 20171022,05:40,1688167,0,
,,17.6,243.4 => OK (200)
[05:45:48] INFO: Uploading 2 datapoints, starting with 20171022,05:40,1785710,0,
,,18,243.01 => OK (200)
[05:50:39] ERROR: Uploading datapoint: 20171022,05:50,1688178,20258148,,,19.6,24
3.64 Bad request 400: Power value [20258148] too high for system size [3328]**_

Herewith the last two lines from my log file.
**_[12:44:39] ERROR: Uploading 82 datapoints, starting with 20171022,05:50,1688178,
20258148,,,19.6,243.64 Bad request 400: Power value [20258148] too high for syst
em size [3520]
[12:44:48] ERROR: Uploading 82 datapoints, starting with 20171022,05:50,1785720,
21428652,,,19.7,243.34 Bad request 400: Power value [21428652] too high for syst
em size [3655]_**

The capacity for the strings in PVOutput are:   3655 Watt  &  3520 Watt

05h50 in the morning only 120Watt was generated (About the same as in the passed)

I also rebooted the Pi just in case - stays the same.

Some pointers would be appreciated.

Regards

Frits

sbf-, 2017-10-22:
@fritserasmus See [this thread](https://sbfspot.codeplex.com/discussions/653932) for solution

sbf-, 2017-10-22:
To avoid this in the future, use [this workaround](https://sbfspot.codeplex.com/workitem/132)

fritserasmus, 2017-10-28:
@sbf- SBF
 
Finally I got time to try this but I was not successful.
I am a Noob, so please be patient with my stupid questions~

The database name and location:  

pi@pi://home/pi/smadata $ Oct 28 18:00 SBFspot.db

from command line on the RasberryPi, 
1) How do I OPEN the database
2) Will -     select * from daydata where power>20200000;     list the faulty records?   
                Just a check I am in the correct table?
3) I understand from there I should execute the commands:
            update daydata set power=0 where power=20258148;
            update daydata set power=0 where power=21428652;

The workaround:

Do I understand correct: vwPvoData is a table in the SBFspot.db 
                                      Should I then execute the instructions almost on the end of the page starting with this:

DROP VIEW IF EXISTS vwPvoData;

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              CASE WHEN dd.Power > 1800
              THEN 0
              ELSE dd.Power
              END AS V2,
..................?

Thanking in advance

Noob: Frits

sbf-, 2017-10-28:
All the things you say are correct
Open db:
`sqlite3` /home/pi/smadata/SBFspot.db`
Check the faulty records:
`select * from daydata where power>20000;`
If you find the records, delete them:
`update daydata set power=0 where power>20000;`
Now the daemon should start uploading...

Workaround is OK, replace 1800 with 150% of the size of your system.

Hope this helps

fritserasmus, 2017-10-28:
FOUND the solution:

Configured SMB server on Pi
     https://raspberrypihq.com/how-to-share-a-folder-with-a-windows-computer-from-a-raspberry-pi/
         **Just do an update sudo update first**

Installed SQLiteStudio on windows
        Connected via gui and easy

fritserasmus, 2017-10-28:
SBF:
I would like to update the vwPvoData View. Below is my current View. Since I have two inverters I would like to make 100% sure I do the correct action, could you please advise if the text below can be executed after DROPping the original file:

CREATE VIEW vwPvoData AS
      SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              dd.Power AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;

As per bottom of: 
https://sbfspot.codeplex.com/workitem/132

Regards

Frits

sbf-, 2017-10-29:
Sorry, but I don't really get what you want to do...

sbf-, 2017-11-21:
No more FB received - closing

closed 2017-11-21 by sbf-