SBFspot/SBFspot #5: Installation sbfspot v.331 Warnings

blindfoxx, 2017-10-24:


blindfoxx, 2017-10-24:
Hi sbfspot-team,
at first: thank u for u're awesome work. 
I've installed sbfspot v.3.3.1 at my fathers raspberryPi V2 with raspbian-strech.
There were the following warnings during the installation:
```
pi@hetke_pv_raspberrypi:~/sbfspot.3/SBFspot $ sudo make install_sqlite
test -d bin/Release_SQLite || mkdir -p bin/Release_SQLite
test -d obj/Release_SQLite || mkdir -p obj/Release_SQLite
g++ -Wall -O2 -DUSE_SQLITE  -c boost_ext.cpp -o obj/Release_SQLite/boost_ext.o
g++ -Wall -O2 -DUSE_SQLITE  -c db_SQLite.cpp -o obj/Release_SQLite/db_SQLite.o
g++ -Wall -O2 -DUSE_SQLITE  -c db_SQLite_Export.cpp -o obj/Release_SQLite/db_SQLite_Export.o
g++ -Wall -O2 -DUSE_SQLITE  -c misc.cpp -o obj/Release_SQLite/misc.o
g++ -Wall -O2 -DUSE_SQLITE  -c strptime.cpp -o obj/Release_SQLite/strptime.o
g++ -Wall -O2 -DUSE_SQLITE  -c sunrise_sunset.cpp -o obj/Release_SQLite/sunrise_sunset.o
g++ -Wall -O2 -DUSE_SQLITE  -c SBFNet.cpp -o obj/Release_SQLite/SBFNet.o
g++ -Wall -O2 -DUSE_SQLITE  -c Bluetooth.cpp -o obj/Release_SQLite/Bluetooth.o
g++ -Wall -O2 -DUSE_SQLITE  -c CSVexport.cpp -o obj/Release_SQLite/CSVexport.o
g++ -Wall -O2 -DUSE_SQLITE  -c Ethernet.cpp -o obj/Release_SQLite/Ethernet.o
g++ -Wall -O2 -DUSE_SQLITE  -c EventData.cpp -o obj/Release_SQLite/EventData.o
g++ -Wall -O2 -DUSE_SQLITE  -c ArchData.cpp -o obj/Release_SQLite/ArchData.o
g++ -Wall -O2 -DUSE_SQLITE  -c SBFspot.cpp -o obj/Release_SQLite/SBFspot.o
SBFspot.cpp: In function ‘E_SBFSPOT initialiseSMAConnection(InverterData*)’:
SBFspot.cpp:1408:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
     if (DEBUG_NORMAL)
     ^~
SBFspot.cpp:1413:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
  do
  ^~
SBFspot.cpp: In function ‘int GetConfig(Config*)’:
SBFspot.cpp:2457:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
     if (strlen(cfg->plantname) == 0)
     ^~
SBFspot.cpp:2460:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
  if (cfg->timezone.empty())
  ^~
SBFspot.cpp: In function ‘E_SBFSPOT setDeviceData(InverterData*, LriDef, uint16_t, Rec40S32&)’:
SBFspot.cpp:3252:5: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
     else
     ^~~~
SBFspot.cpp:3255:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘els ’
  return rc;
  ^~~~~~
SBFspot.cpp: In function ‘E_SBFSPOT getDeviceData(InverterData*, LriDef, uint16_t, Rec40S32&)’:
SBFspot.cpp:3280:5: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
     else
     ^~~~
SBFspot.cpp:3283:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘els ’
  int validPcktID = 0;
  ^~~
g++ -Wall -O2 -DUSE_SQLITE  -c TagDefs.cpp -o obj/Release_SQLite/TagDefs.o
g++  -o bin/Release_SQLite/SBFspot obj/Release_SQLite/boost_ext.o obj/Release_SQLite/db_SQLite.o obj/Release_SQLite/db_SQLite_Export.o obj/Release_SQLite/misc.o obj/Release_SQLite/strptime.o obj/Release_SQLite/sunrise_sunset.o obj/Release_SQLite/SBFNet.o obj/Release_SQLite/Bluetooth.o obj/Release_SQLite/CSVexport.o obj/Release_SQLite/Ethernet.o obj/Release_SQLite/EventData.o obj/Release_SQLite/ArchData.o obj/Release_SQLite/SBFspot.o obj/Release_SQLite/TagDefs.o   -s -lbluetooth -lboost_date_time -lboost_system -lpthread -lsqlite3
cp TagList*.txt bin/Release_SQLite
cp date_time_zonespec.csv bin/Release_SQLite
test -d /usr/local/bin/sbfspot.3 || mkdir -p /usr/local/bin/sbfspot.3
test -f /usr/local/bin/sbfspot.3/SBFspot.cfg || cp SBFspot.cfg /usr/local/bin/sbfspot.3/
cp -R bin/Release_SQLite/* /usr/local/bin/sbfspot.3/
pi@hetke_pv_raspberrypi:~/sbfspot.3/SBFspot $
````
Should I be worried?

sbf-, 2017-10-24:
No problem to ignore these warnings... I didn't compile it on Stretch until now.
See also [issue 160](https://sbfspot.codeplex.com/workitem/160)

blindfoxx, 2017-10-25:
Ok,
thank you.

closed 2017-10-25 by blindfoxx