SBFspot/SBFspot #6: Failed to get current plant time (-1)

rarakers, 2017-10-28:
Hi,

I get the error message above when executing SBFspot 3.3.1. Interestingly the first time I executed SBFspot there was no such issue:
https://pastebin.com/PnYZnTcb

From now on I get the current output:
https://pastebin.com/Z0UCZb0D

Here is another log with "-d5 -v5" options:
https://pastebin.com/X3kF8Wbb

So actually a timeout occurs when SBFspot tries to read the current plant time. This didn't happen on the first time starting SBFspot.

Can you help me? Thanks!

sbf-, 2017-10-28:
The time of your inverter is more than 3600 sec out of sync (05/03/2005 04:43:26) so SBFspot doesn't set it automatically.
Try with `SBFspot -settime` to adjust it for the first time (or with Sunny Explorer on windows)
It is in the first log:
`Local Host Time: 28/10/2017 11:18:43 `
`Plant Time     : 05/03/2005 03:59:35 (-399190748 sec)     `
`TZ offset      : 3600 sec - DST: On     `
`Last Time Set  : 09/07/2013 13:06:08 `    
`The time difference between inverter/host is more than 3600 seconds.     `
`As a precaution the plant time won't be adjusted.     `
`To overrule this behaviour, execute SBFspot -settime `

rarakers, 2017-10-28:
Thanks for your response. I already tried to synchronize the time as you described but it doesn't work.
Also I would expect to see the time difference message all the time, but instead retrieving the inverter date time results in an error since running SBFspot for the first time.

sbf-, 2017-11-03:
I suppose it works with Sunny Explorer (Windows only)?

rarakers, 2017-11-04:
Unfortunately my BT stick is not supported by Sunny Explorer. I'll try a different one and report back next week when it arrives.

npkamen, 2018-03-28:
Did you ever get a resolution on this?  I am having the same trouble and can't get the data out of the inverter.  Trying to set the time fails and it won't fetch the data without having the time set.

rarakers, 2018-03-28:
The Bluetooth modules I tried didn’t work with Sunny Explorer. So no resolution achieved until now. I deactivated the time setting in the configuration file for a (permanent) workaround I guess.

sbf-, 2018-03-28:
@rakete1 but if inverter time is incorrect, SBFspot will not work as it should...
FTM I have no solution for this issue, but I can give it another try.
Can someone send me a debug log?
`SBFspot -v5 -d5 -settime>debuglog.txt`
Please send it to s.b.f (at) skynet.be

sbf-, 2018-03-28:
Reported by @rakete1 (e-mail)

> Hi,
lol.. now it seems to work?! I tried it twice and in the first time it sets the new time successfully.. The second execution detected a time offset of 0 so I’d say it successfully set the time in the first place.
BR
Ralph

sbf-, 2018-03-28:
That's why it makes it so difficult to fix.

Doudy, 2018-08-12:
Hi,
I have the same problem with SBFspot V3.5.0

pi@raspberrypi78 /usr/local/bin/sbfspot.3 $ ./SBFspot -settime

SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -settime
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Sun Aug 12 09:04:01 2018: INFO: Starting...
sunrise: 06:29
sunset : 21:10
Connecting to 00:80:25:26:F9:49 (1/10)
Initializing...
SUSyID: 125 - SessionID: 927082544 (0x37422830)
SMA netID=05
Serial Nr: 77577C58 (2002222168)
BT Signal=72.9%
Logon OK
Failed to get current plant time (-1)

Doudy, 2018-08-12:
Notice that
When i do 
/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv => no problem
When i do 
/usr/local/bin/sbfspot.3/daydata => SetPlantTime returned an error: -1
! For test I have changed daydata with
/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv >> $log

What does it mean ?

AJErazzor, 2018-10-26:
oddly although I have this error I have data in the database whenever it runs - nothing on pvoutput however?

sbf-, 2018-10-26:
You should check the time of your inverter and in the database (vwPvoData view)
If needed you'll have to adjust it with SunnyExplorer as SBFspot fails to do it

MarkusBe, 2019-05-15:
I have kind of the same error, but with many the functions. I get for almost every query the result "returned an error: -1"

sudo /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Wed May 15 22:35:40 2019: INFO: Starting...
sunrise: 05:37
sunset : 21:16
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 912411879 (0x36624CE7)
Inverter IP address: 192.168.178.73 from SBFspot.cfg
Logon OK
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1
getTypeLabel returned an error: -1
getDeviceStatus returned an error: -1
getInverterTemperature returned an error: -1
getMaxACPower returned an error: -1
getEnergyProduction returned an error: -1
getOperationTime returned an error: -1
getSpotDCPower returned an error: -1
getSpotDCVoltage returned an error: -1
SUSyID: 408 - SN: 3003926934
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
getSpotACPower returned an error: -1
getSpotACVoltage returned an error: -1
getSpotACTotalPower returned an error: -1
SUSyID: 408 - SN: 3003926934
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.000kW
getSpotGridFrequency returned an error: -1
********************
* ArchiveDayData() *
********************
startTime = 5CDB3A60 -> 15/05/2019 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 5CC97C30 -> 01/05/2019 13:00:00
Reading events: 2019-May-01
Wed May 15 22:36:50 2019: INFO: Done.

How can I fix this?

sbf-, 2019-05-15:
@MarkusBe This issue is fixed in 3.5.1

MarkusBe, 2019-05-20:
Thanks, I reinstalled the whole thing, now it works like a Charm!

closed 2020-06-24 by sbf-