SBFspot/SBFspot #137: workaround for: Power value too high for system size

SBFspot, 2017-11-04:
Hello,

I recently started using SBFspot, version 3.1.1 on raspberry pi.

I tried to get all archived data from the SMA inverter,
and then upload it to pvoutput.org.

I used the -ad and -ae flags too fetch the archived data
(I used -ad300 and -am300, which are larger than the
'live' data that is present in the inverter -- it gave me
'live' data for a few months;
it gave me  per-day totals for all months since the inverter was activated.

Somehow, the very first datapoint for the very first day of the archived data
has an 'interesting' power/energy (V2) value:
I assume it is the amount of energy generated  from the moment the SMA inverter was enabled upto that moment in time.
In my case, this is a large number:  110925396

In the csv file it looks as follows:
 12/07/2015 04:25:00;9243,782;110925,396
 12/07/2015 04:30:00;9243,782;0,000

When uploading the archived data to pvoutput.org
SBFspot tries to upload this huge value as a value for V2.
Pvoutput.org refuses the upload, because the value is higher than the system size:
Bad request 400: Power value [110925396] too high for system size [3825]

The following patch kind-of fixes this:
when obtaining the archived data using batch_get_archdaydata we take
the system size into account: the system size is passed as additional
parameter in the call to batch_get_archdaydata.

When it encounters a V2 value that is higher than system size,
it will use 0 instead; when that happens, a LOG line is printed.

This works for me; I only tested this with mysqlite.

Note: I slightly edited the attached diff to remove irrelevant changes;
because of this, line numbers in the patch may be slightly off.

Axel.
PS thanks for SBFspot, it does a nice job! :-)

#### Attachments
[too-high-V2.diff](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=132&FileAttachmentId=1484847)
[log-high-val-ou.txt](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=132&FileAttachmentId=1490466)
[SBFspotUpload20160821.log](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=132&FileAttachmentId=1602363)

#### This work item was migrated from CodePlex
CodePlex work item ID: '132'
Vote count: '1'

SBFspot, 2017-11-04:
[AxelBelinfante@15/09/2015]
I just looked at my logs, it contains a line:
```
WARNING: Passing too high V2 value as 0: V2=110925396 > systemSize=3825
```

A minor improvement to my patch would be to include date-time information of the data point in the log message.

Axel.

SBFspot, 2017-11-04:
[SillieWimons@18/09/2015]
dear axel,   in sequence my comments on your information

-a- you did identify an issue for sbfspot,   IT WAS NOT PRESENT IN PREVIOUS VERSIONS
NEW ISSUE:
if the maximum archive of daily information is retrieved then the first value is not treated correctly
the high value 110925396 is ~the result of the calculation 

-b- to allow a complete analysis, can you provide the two log files using the below "extra" options:
extra means supplement to the options your using
sbfspot -d5 -v5 1>log-high-val-ou.txt 2>log-high-val-er.txt
rational-1:
the high value is not a perfect match to the calculatuion (there is a delta of ,012)
rational-2:
the log file will help to control the information given by your inverter,

FOR YOUR INOFORMATION: most inverters will deliver via archive-requests 
-am###  a complete list of month files
-ad###  about 61days of daily files
-ae###  about 250events (user) and 250events(installer)

kr wim

SBFspot, 2017-11-04:
[SillieWimons@23/09/2015]
dear axel,  

after doing a full search on my envirnment,  i found that your issue does occur also with v308
BUT it is not persistent ->  it appeared "only twice in 18" executions

can you please forward the additional information as requested above
I expect that you can modify your patch such that at occurence of the issue 
you can generate the additional information
 
kr wim

SBFspot, 2017-11-04:
[AxelBelinfante@24/09/2015]
submit to codeplex just failed, and I lost  the long piece that  I typed :-(

I am confused by what you ask, because I would guess that 

```
sbfspot -d5 -v5 
```
only gives info about fetching 'current' data from the convertor;
maybe I misunderstood what you are asking for; I attach the stdout output anyway; the stderr output is empty.

Sidenotes:
 1) the sbfspot I used to obtain attached output differs from the one I used to obtain the archived data from the convertor:
I compiled it with a newer version of bluez that I installed from source.
Reason: initially I had installed bluez  using apt-get; with that version, after a while (from 12 hours upto several days?) bluetooth connections were no longer working/possible.

2) my convertor clock is approx 6 minutes ahead of the (ntp-controlled, I think) computer clock.
timeSync is enabled but does not seem to have any effect.
Have you seen timeSync work?

3) voltage/temperature data appears only on pvoutput when i restart the raspi in the evening
when bluetooth connection stopped working during the day.

one day this gave me weird results:
http://pvoutput.org/intraday.jsp?id=44413&sid=40547&dt=20150918&gs=0&m=0

regards,
Axel.

SBFspot, 2017-11-04:
[SillieWimons@28/09/2015]
dear sbf   and  axel

@sbf   allow me to propose to insert a WARNING when
the calculation for the "production", in the daily table, is above 20.000Watt

rational-1:
I expect this limit to be above the Max-power for any single inverter handled via SBFspot
note: an alternate is to read the inverters max power and apply a risc factor of 1.5
that alternate method might be needed for small converters, SB240

rational-2:
that warning is valid once an inverter has a total production larger then 1.666kWh
only for SB240 the limit of 20kW might not fit practical usage

@axel - your point about the option -d5 -v5      and the redirect
thanks for this remark - we  were wondereing wij one needs to ask this - each time again
the options -d5 -v5 instruct sbfspot to write a maximum of information into the log file
and these options apply to "all types of functions" part of sbfspot
the log file is usefull since:
- it does include the "data-buffers" for the interaction vav inverter
- it includes messages (warning, error, . . . ) from sbfspot

@axel - your-point-2 - there is an issue about the time-sync,
in earlier versions of sbfspot that function did work - the point is under investigation

kr wim

SBFspot, 2017-11-04:
[SillieWimons@29/09/2015]
dear axel,    i did look in your log file, 

since issue C132 is expected to appear if you request the options -ad300 (apply at least -ad80)

may i ask you to repeat the test (if possible with v312) 
using the options: sbfspot -ad80 -d5 -v5   and   redirect the log file

thank you for you cooperation   wim

SBFspot, 2017-11-04:
[ultraschall81@1/06/2016]
Hi,

I'm facing the same problem (around once a week/every 3-4 Days). It looks like it happens always when there is very little solar power.
I use a PI with SBFSpot 3.1.5 (also tried 3.1.2) and a SMA SB1300.
Before the upgrade I used SBFSpot 2.x and it was working fine.

Currently there is one dataset with a high value in my SQLiteDB, which I will keep in case you will request for the dbfile.
Normaly I use a DB-Browser to change the wrong value to "0" and the upload is fine again.

BTW: I posted this in the closed issue 109 as well.

Br
Christian

SBFspot, 2017-11-04:
[SBF@2/06/2016]
PVoutput allows a max power value=150% of the system size. In your case, for an SB1300 this is appox. 1950W

You can change the vwPvodata view to limit the powervalue to 1800 (to be on the safe side)
```SQL
CASE WHEN dd.Power > 1800
    THEN 0
    ELSE dd.Power
END
```

The comlete view looks like this: (SQLite has no "ALTER VIEW" so you have to DROP it first.)
```SQL
DROP VIEW IF EXISTS vwPvoData;

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              CASE WHEN dd.Power > 1800
              THEN 0
              ELSE dd.Power
              END AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;
```

SBFspot, 2017-11-04:
[ultraschall81@3/06/2016]
Thanks for your help. Looks like it is working.

Br

Christian

SBFspot, 2017-11-04:
[simonbeswick@21/08/2016]
I am receiving the error whilst running the latest version of SBFspot could someone please advise how I resolve this ?

Thanks

SBFspot, 2017-11-04:
[simonbeswick@21/08/2016]
Log file in case it helps

SBFspot, 2017-11-04:
[DaveDE@12/10/2016]
Hello, I have the same problem described above and would like to add the Case statement to prevent future errors. I'm a rank Linux newby and do not know how to modify the file. What file am I modifying? I can not find the file that has the vwPvoData. Please explain in a little more detail how to add the case statement ie; what directory I need to be in, what commands are necessary to bring the file up, etc. Thanks! Dave

SBFspot, 2017-11-04:
[DaveDE@12/10/2016]
...looks like I have to edit the SQLite database CREATE VIEW vwPvoData. I'm using DB Browser for SQLite but don't see how to edit the VIEW. Do I need another editor? Thanks.

SBFspot, 2017-11-04:
[DaveDE@12/10/2016]
Never mind, got it figured out. Like you said, the vwPvoData view has to be deleted and a new one created with the case statement inserted. DB Browser can do this using the Execute SQL tab.

It would be nice if future versions on SBFSpot has the case statement in there with a default of say 1,000,000 watts. That would fix almost all errors since they seem to always be astronomically high values. Thanks, this program is great!

SBFspot, 2017-11-04:
[Atlantis69@19/10/2016]
Just adding to the thread.

I set up my logging on a Windows machine about a week ago. Perfect logging every 5 minutes... until yesterday when this occured as well. As others have done, I went in with DB Browser and fixed the data.
I had a look at the log CSV files and can't see any issues with the data in there (unless I'm not looking correctly), so I have a feeling SBF is misinterpreting the data when it's being placed into the SQL DB.
Quite happy to provide logs if anyone needs to have a look.

drdiesel, 2018-10-04:
I was running into this same problem every few days, just made the view modification above, thanks!

goretex1973, 2018-11-12:
Hi

I am running the latest version 3.5 and I am getting the error 

ERROR: Uploading 100 datapoints, starting with 20181107,14:35,20011128,204,,,47.6,245.65 Bad request 400: Power value [2147483647] too high for system size [36000]

Can some one please provide some assistance in resolving the error

Thanks

sbf-, 2018-11-14:
You have to modify the view vwPvoData....

PVoutput allows a max power value=150% of the system size. In your case, for a 36kW system (are you sure about that?) this is 54000W

You can change the vwPvodata view to limit the powervalue to 50000 (to be on the safe side)
```SQL
CASE WHEN dd.Power > 50000
    THEN 0
    ELSE dd.Power
END
```

The comlete view looks like this: (SQLite has no "ALTER VIEW" so you have to DROP it first.)
```SQL
DROP VIEW IF EXISTS vwPvoData;

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              CASE WHEN dd.Power > 50000
              THEN 0
              ELSE dd.Power
              END AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;
```

fritserasmus, 2019-07-29:
.fullschema shows:

```
CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              dd.Power AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
```
When I try to DROP the view:- This is what I get

pi@SMA2PVOUTPUT:~/smadata $ DROP VIEW IF EXISTS vwPvoData;
-bash: DROP: command not found

Any help?

scotte, 2019-07-29:
That's a sqlite command, not shell - run it in sqlite after you've connected to the schema.

fritserasmus, 2019-07-29:
Thanks,

Figured it out in the end.

Thanks for the help

snowlucas, 2019-08-21:
HI,

I have this issue so I followed the work around and created a view as below. It works for the rest of the day and then as soon as I start production in the morning I get the errors back. I believe a reboot will get it working again for the rest of the day.

Any ideas on how I can get this working properly?

The inverter model is SMA SB3.6-1AV-40 558

Thanks

[09:51:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:52:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:53:32] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:54:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:55:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:56:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:57:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]
[09:58:31] ERROR: Uploading 100 datapoints, starting with 20190612,16:45,5833020,69996252 Bad request 400: Power value [69996252] too high for system size [4960]

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              CASE WHEN dd.Power > 7500
              THEN 0
              ELSE dd.Power
              END AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;

sbf-, 2019-08-21:
Very strange if you ask  me...
What is the output of
`select * from vwPvoData;`
Maybe restart daemon?

snowlucas, 2019-08-21:
sqlite> select * from vwPvoData;
Error: no such table: main.vwDayData

snowlucas, 2019-08-21:
I used the sbfspot-config you provided to create everything.

Daydata looks like this 

#!/bin/bash
#
log=/var/log/sbfspot.3/Forest_Road_$(date '+%Y%m%d').log
/usr/local/bin/sbfspot.3/SBFspot -v -ad1 -am0 -ae0 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log

sbf-, 2019-08-21:
It looks like you deleted the daydata view. Recreate it with:

``` SQL
CREATE View vwDayData AS
	select datetime(Dat.TimeStamp, 'unixepoch', 'localtime') AS TimeStamp,
	Inv.Name, Inv.Type, Dat.Serial,
	TotalYield,
	Power,
	PVOutput FROM DayData Dat
INNER JOIN Inverters Inv ON Dat.Serial=Inv.Serial
ORDER BY Dat.Timestamp Desc;
```

snowlucas, 2019-08-23:
Tried this and it continued to say the same no such table error. 

I removed everything. Followed the full install instructions rather than use the install script and it now seems to be working without any tweaks. 

Thanks for your help.

saylinglow, 2020-01-16:
I've been running SBFspot 3.5.0 on a Raspberry Pi and I found that the Power value is miscalculated if I change the total generation (kWh) setting in the Sunny Boy inverter. I do this periodically because the total generation value in the Sunny Boy inverter drifts (advances) relative to the separate generation power meter I use to report my output to my electricity supplier. When I set the Sunny Boy total generation value back by the amount it has drifted to make it the same as the power meter, the value placed in DayData.Power in the SQLite database is a nonsense extremely high value because the energy difference is negative. 

Would it be possible to change the existing test in the procedure ArchiveDayData in the file ArchData.cpp (line 149 in the v3.6.0 source)? :
from
+++
			if (totalWh > 0) hasData = E_OK;
			if (totalWh_prev != 0)
			{
+++
to
+++
			if (totalWh > 0) hasData = E_OK;
			if ((totalWh_prev != 0) && (totalWh >= totalWh_prev))
			{
+++

The changes to vwDayData are fine, but we should be trying to stop the data being corruptly entered in the database to start with.

best wishes
Sayling Low

Depechie, 2020-03-26:
Hey... I'm also getting : Bad request 400: Power value [2147483647] too high for system size [4600]

But sorry, being a noob with Linux and SqLite, can someone tell me a bit in detail the command to do the work around?
Because if I go into the sqlite terminal on my raspberry pi : sqlite3 SBFspot
Doing the .tables returns nothing and also doing a select : select * from vwDayData; returns No such table vwDayData.

So in other words a bit stuck on what to do?

sbf-, 2020-03-26:
You have to provide the full path:
sqlite3 /home/pi/smadata/SBFspot.db

Depechie, 2020-03-26:
I was in that directory before I did the sqlite3 command.
I just seem to see the same problem as snowlucas had. Nothing comes up when I try sql commands in the sqlite3 terminal.

sqlite> select * from vwPvoData;
Gives
Error: no such table: main.vwDayData

After trying to recreate it but uploading is not restarting.
What I meant about Linux was that I wanted to copy over the sqlite db file to my Mac to get a look at it with a db viewer. But not able to get that done.

So what is the best course of action? Try the manual install like snowlucas?

sbf-, 2020-03-27:
It looks like you have overwritten the db with an empty one. If you don't have a backup you have to create a new one.

Depechie, 2020-03-27:
Hey @sbf- I was able to recreate the DB and select * from vwspotdata; does return data from today!
Is there an easy way to import the missing CSV data? ( it is still available in the subdir 2020 ).

sbf-, 2020-03-27:
Importing csv files is not yet possible.
You can import 60 days of historical data, spotdata is lost.
SBFspot -ad60 -am60 -ae60 -nocsv -finq

Depechie, 2020-03-27:
Ok thx! maybe 1 more Q, after I recreated the DB and restarted the daemon I only get following in the log file of today, so nothing is being uploaded. Any clue why that could be?

[10:18:40] INFO: SBFspotUploadDaemon Version 2.0.0
[10:18:40] INFO: Starting Daemon...

sbf-, 2020-03-27:
The actual logging is in /var/logs/sbfspot.3 or smadata/logs

Depechie, 2020-03-27:
Thx for the responses by the way.

True... the file SBFspotUpload20200327.log inside /var/logs/sbfspot.3 only contains those 2 lines for today.
[10:18:40] INFO: SBFspotUploadDaemon Version 2.0.0
[10:18:40] INFO: Starting Daemon...

Normally it should be filled with the uploading datapoints ( file of 25th has that, 26th has that up until the exception of the size )

The new sqlite db does contain data of today, so the bluetooth connection to my sma is fine I guess ( I also see data in the csv file of today in the 2020 folder ).

So what am I missing..., sorry don't know what info I can provide to solve this.

sbf-, 2020-03-27:
Have a look at #370

Depechie, 2020-03-27:
@sbf- THANKS! So I did have one 1 view not correctly configured inside the DB. Now everything works again.
Thanks for your time.

Driekes, 2020-04-16:
I have the same issue with my Sunny Tripower 5.0. This happened two times in the first week. If I remove the first value of the day it is working correctly.

[06:35:31] ERROR: Uploading 2 datapoints, starting with 20200416,01:30,214298,0 Bad request 400: Power value [2571588] too high for system size [5760]

I'm running on Windows so I don't think I'm able to change the code. Do you have any advice?

sbf-, 2020-04-16:
Windows or Linux, the procedure is the same: alter the view as described in my post of november 14th, 2018 in this thread.

Driekes, 2020-04-16:
I am using it with MySQL, does that view apply to that? I didn't recognise that command, so wonder if that is applicable?

Driekes, 2020-04-16:
I didn't know of views, but I found them. The underlying data might still be 'wrong', but the view now translates to data so that PVOutput accepts it, correct?

sbf-, 2020-04-16:
> I didn't know of views, but I found them. The underlying data might still be 'wrong', but the view now translates to data so that PVOutput accepts it, correct?

Correct

hollie, 2020-04-27:
I can confirm I hit the same issue with an old SB3000-TL20. The first reading of the day was incorrect, causing PVOutput to reject the input from that point onwards.

In my case the power value was high, but not the high numbers I see others reporting:

```
Bad request 400: Power value [14556] too high for system size [3150]
```

I have updated the view to filter out values higher than 4000 and now all is working again as expected.

@sbf- Thanks for the support and thank you for developing SBFspot and making it available to us!

snoekieboe, 2021-01-24:
Still hitting this issue even on a brand new and most recent install. Is this being worked on?

sbf-, 2021-01-24:
Because of the available workaround, this is low priority.

roelvdh, 2021-02-12:
> The actual logging is in /var/logs/sbfspot.3 or smadata/logs

Mine is at /var/log/sbfspot.3, not /var/logs/sbfspot.3

roelvdh, 2021-02-12:
> [SBF@2/06/2016]
> PVoutput allows a max power value=150% of the system size. In your case, for an SB1300 this is appox. 1950W
>
> You can change the vwPvodata view to limit the powervalue to 1800 (to be [...]

In case you have 2 systems the number entered should not exceed 150% of either system size.

sbf-, 2021-02-12:
@roelvdh Right, I corrected the typo. Thanks.

sbf-, 2021-02-12:
> In case you have 2 systems the number entered should not exceed 150% of either system size.

That's also true... I'm working on a better solution

danstreeter, 2021-11-06:
Thanks for the workaround - I got hit by this today after leaving my install isolated/turned off overnight. Seems a strange value came through for some reason.

Performing the drop and re-create worked lovely.

closed 2021-12-11 by sbf-

madtempest, 2022-03-27:
Just as an FYI, this will also happen if you screw up your panel count vs watts per panel.

I just found the error myself which looks like this, but in fact the limitation was in PVOutput.orgs data through my error, not the script.

HeinvdW, 2022-12-26:
Solution for this Power value too high for sytem size
cd ~/smadata or cd smadata
sqlite3 SBFspot.db
 select * from daydata where power > 5000;  (5000 maybe 3000, 4000 or whatever is the max of your inverter)
Play a little and you'll end up with just a few lines.
delete from daydata where power > 6000; (or 5000 or 4999 or 3020)
Close sqlite with Ctrl-d

On your RPI terminal:
 /usr/local/bin/sbfspot.3/SBFspot -v -finq -ad60  (ad60 retrieves daydata from up till 60 days back)
You can change this for three day for eamble:  /usr/local/bin/sbfspot.3/SBFspot -v -finq -ad3
Have a few minutes patience and your data will show up in PVOutput.

commit 527b2a4b (2021-12-11): Fix various issues Bad request 400: Power value too high for system size:
Fix #137
Fix #313
Fix #381
Fix #384