SBFspot/SBFspot #148: Keep BT connexion open

SBFspot, 2017-11-04:
Hello, 

I would like to refresh the spotvalues every X seconds. 

Actually, there is a big "time/compute" overhead due to the socket connect/disconnect.

It should be possible to keep a persistent connection to the inverter and so, save this overhead.

It will also say that the process would be "deamonized" ...

Regards,
Laurent.

#### This work item was migrated from CodePlex
CodePlex work item ID: '144'
Vote count: '1'

SBFspot, 2017-11-04:
[WillieSimons@31/01/2016]
dear laurent,  

I include the below information, based on my experience, having used a script to repeatedly 
read the spot values (using the options -ad0 -am0 -ae0)
with such machinegun script i was able to execute 10 readings in <15seconds

For that refresh rate you need to skip the read out of tables for day, month, events, . . .

my experience 
- there is no use for pushing the refresh-interval below 8seconds

- when the interrogation rate gets below 10seconds, then you will see in the answer from your inverter:
-a- the programs, part of your inverter, do not send send the spot values to the communication module
      in a continue mode (transmission from comm's module via Comm's channel to your computer)
-b- if you apply the debug mode to sbfspot (ptions -d5 -v5 ) then you can see in the log-file
      that each spot value has its own time stamp associated
-c- that internal update rate is not fixed (may vary from around 4seconds to 20seconds)
-d- spot values from the DCmodule, ACmodules, . . . in your inverter have different reactions
      
for the above - a referesh rate at >10seconds is valid

 
Keeping the communication channel open in continue mode is a complete new way of interaction,
from observation, using SunExplorer, there are time-outs on the communication channel with the inverter,

An alternate solution for your demand might be - to introduce a variant for the -sp# option:
-sp#   where the # indicates a number of repeats for the spot interogation
          having the refresh interval set by the config file and # coded as
-sp0  <- skip spot values   (actual option)
-sp1  <- execute once  (the current default)
-sp2 <-> 5    read  2 till 5 times in a row the spot values    
         after that repetition - then continue with daily-month-event

kind regards

SBFspot, 2017-11-04:
[WillieSimons@1/02/2016]
dear laurent,   in addition to the above comments

-a- read one session == connect/logon/. . . . . interactions . . . /logoff/disconnect
-b- the data communication towards an inverter (using BT) only supports two parallel sessions,
      additional sessions will be rejected by the inverter,
-c- today - from one computer (RPi or Windows) one can set up ONLY ONE session at a time, 
      this is due to the OperSyst-sockets for BT, 
-d- if you want to extract form your inverter AND spotvalues   AND data for eg PVOutput, then
EITHER you can do this on one computer by running all tasks in serie
note: the above is the preferred setup
OR you must setup two computers - one running a spot values script  - the other for PVoutput
note: with this second option no other apparatus can interact with the inverter

sbf-, 2020-06-10:
See #318

closed 2020-06-10 by sbf-