SBFspot/SBFspot #159: Using 3.3.1 I get "Failed to get current plant time (-1)"

SBFspot, 2017-11-04:
I am using 3.3.1.The first part of the log shows:
Compiled for Linux (LE) 32 bit

Commandline Args: -d1 -v -ad7
Reading config './SBFspot.cfg'
Thu Jul 21 12:05:01 2016: INFO: Starting...
sunrise: 05:47
sunset : 21:46
Connecting to 00:80:25:2C:C5:C7 (1/10)
Initializing...
SUSyID: 125 - SessionID: 970141159 (0x39D32DE7)
SMA netID=02
Serial Nr: 77577C46 (2002222150)
BT Signal=73.7%
Logon OK
Failed to get current plant time (-1)
SetPlantTime returned an error: -1
SUSyID: 99 - SN: 2002222150
Device Name:      SN: 2002222150
Device Class:     PV-omvormers
Device Type:      SB 1600TL-10
Software Version: 12.12.208.R
Serial number:    2002222150
SUSyID: 99 - SN: 2002222150
Device Status:      OK
SUSyID: 99 - SN: 2002222150
Device Temperature: 0.0°C
SUSyID: 99 - SN: 2002222150
GridRelay Status:      ?
SUSyID: 99 - SN: 2002222150
Pac max phase 1: 1600W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 99 - SN: 2002222150
Energy Production:
EToday: 3.702kWh                                                                                                                               
ETotal: 6103.922kWh                                                                                                                            
Operation Time: 15688.51h                                                                                                                      
Feed-In Time  : 12867.38h                                                                                                                      

SUSyID: 99 - SN: 2002222150                                                                                                                        
DC Spot Data:                                                                                                                                          
String 1 Pdc:   0.952kW - Udc: 180.00V - Idc:  5.294A                                                                                          
String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A                                                                                          

SUSyID: 99 - SN: 2002222150
AC Spot Data:
Phase 1 Pac :   0.891kW - Uac: 237.20V - Iac:  3.758A
Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
Total Pac   :   0.891kW

SUSyID: 99 - SN: 2002222150
Grid Freq. : 49.97Hz
SUSyID: 99 - SN: 2002222150
Current Inverter Time: 21/07/2016 12:05:14
Inverter Wake-Up Time: 21/07/2016 12:05:13
Inverter Sleep Time  : 21/07/2016 12:05:14

As you can see I get two error messages about the reading and setting of the planttime.
Later it shows a proper Inverter Time, which is consistent with the computer time

#### This work item was migrated from CodePlex
CodePlex work item ID: '155'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@21/07/2016]
Hi,
can you post (or send by mail) the output with -d5 option?
Thanks

SBFspot, 2017-11-04:
[fdekruijf@22/07/2016]
I did send to you the first 5 parts of the logfile in chunks of 1950 bytes.

npkamen, 2018-03-28:
I am getting this same error.  Any thoughts on the issue?  When I try to issue a -settime option to the script it simply fails to set the time.

closed 2018-09-16 by sbf-