SBFspot/SBFspot #161: Memory leak in SBFspot (all versions)

SBFspot, 2017-11-04:
When using SBFspot with a poor Bluetooth connection to the inverter, memory leaks may occur.
From time to time the BT connection times-out and gets disconnected, causing the program to hang. After some time you'll see several SBFspot processes eating up the memory:
pi@rpi01:~ $ ps -aux|grep SBFspot
root       548 69.4  8.2  52388 36512 ?        Rs   Jul25 8002:23 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon
pi        2120  0.0  0.2   1884  1084 ?        Ss   Jul29   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi        2121  0.0  0.5   5140  2476 ?        S    Jul29   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi        2127  0.0  0.7   5496  3404 ?        S    Jul29   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi        7192  0.0  0.2   1884  1096 ?        Ss   Jul26   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi        7193  0.0  0.5   5140  2492 ?        S    Jul26   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi        7199  0.0  0.7   5496  3524 ?        S    Jul26   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi       10272  0.0  0.2   1884  1092 ?        Ss   Jul30   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi       10274  0.0  0.5   5140  2512 ?        S    Jul30   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi       10285  0.0  0.8   5496  3632 ?        S    Jul30   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi       10882  0.0  0.4   4804  1784 pts/0    S+   23:10   0:00 grep --color=auto SBFspot
pi       11312  0.0  0.2   1884  1084 ?        Ss   Jul26   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi       11313  0.0  0.5   5140  2492 ?        S    Jul26   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi       11319  0.0  0.8   5496  3640 ?        S    Jul26   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi       12770  0.0  0.2   1884  1092 ?        Ss   Jul30   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi       12771  0.0  0.5   5140  2564 ?        S    Jul30   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi       12777  0.0  0.8   5496  3664 ?        S    Jul30   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi       13867  0.0  0.2   1884  1068 ?        Ss   Jul30   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi       13868  0.0  0.5   5140  2520 ?        S    Jul30   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi       13874  0.0  0.7   5496  3556 ?        S    Jul30   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv
pi       14979  0.0  0.2   1884  1112 ?        Ss   Aug01   0:00 /bin/sh -c /usr/local/bin/sbfspot.3/SBFspot.sh > /dev/null
pi       14980  0.0  0.5   5140  2540 ?        S    Aug01   0:00 /bin/bash /usr/local/bin/sbfspot.3/SBFspot.sh
pi       14986  0.0  0.7   5496  3528 ?        S    Aug01   0:00 /usr/local/bin/sbfspot.3/SBFspot -installer -password:1111 -v5 -ad1 -am1 -ae1 -nocsv

When we look at the log of Aug01, we can see this:
Energy Production:
EToday: 0.852kWh
ETotal: 22764.312kWh
Operation Time: 24210.94h
Feed-In Time  : 23521.01h

getSpotDCPower returned an error: -1
getSpotDCVoltage returned an error: -1
SUSyID: 78 - SN: 2100XXXXX
DC Spot Data:
String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A

getSpotACPower returned an error: -1
recv() returned an error: 110
getSpotACVoltage returned an error: -1
send() returned an error: 107
Mon  1 Aug 09:45:01 CEST 2016

All other log files have the same socket errors:
Error 110 = Connection timed out
Error 107 = Transport endpoint is not connected

#### This work item was migrated from CodePlex
CodePlex work item ID: '157'
Vote count: '2'

freekdk, 2018-06-16:
I noticed the same problem of many active SBFspot processes. I used to use 'timeout 230s ' in front the activation of SBFspot, however that did not prevent the many processes. So now I am using 'timeout -k 30s 230s '.

mr-russ, 2019-09-27:
Is anyone able to explain why an upload daemon is used rather than just making the uploader a scheduled task? Both windows and linux support them, unlike services they don't require elevated privileges and is reduces the impact of memory leak issues.

sbf-, 2019-09-27:
This is a design choice I took several years ago and it works very well I think.
This memory leak issue is in SBFspot when using Bluetooth connection. The uploader is not impacted.

MatteoVentura, 2021-02-22:
I am having the same issue, Is this issue resolved in the latest builds? Or what is the advised workaround?

MatteoVentura, 2021-02-24:
It looks like the processes get stuck in state D probably waiting for some Bluetooth I/O data. It's really annoying cause I can't even kill the stuck processes manually or with a timeout.
@sbf- 
Can't a timeout be programmed when trying to read the Bluetooth data?

sbf-, 2021-02-25:
No, still not fixed. Since I moved the Pi closer to the inverter it never happened again.
Strange you can't kill the process.

There is a timeout, but I think it is not catched.
```
recv() returned an error: 110
Error 110 = Connection timed out
```
I'll have to look at it

mcc05, 2022-12-19:
Think I have a similar issue, using the windows version 3.9.6. It runs well for a while (every min), then, goes into a state of failing to connect retry etc, but never recovers. Only way to get it working again is to kill the process and disable/re-enable the bluetooth device in device manager and the start the process again and its all good. Suggests a memory leak?

wrouesnel, 2023-06-11:
I can confirm this happens every morning for me: when the inverter wakes up just enough for Bluetooth to connect, (or maybe in the wrong part of the login cycle?) but the sun isn't fully shining yet I see this in the logs:

```
Jun 10 17:02:41 sbfspot SBFspot[159557]: SBFspot V3.9.7
Jun 10 17:02:41 sbfspot SBFspot[159557]: Yet another tool to read power production of SMA solar inverters
Jun 10 17:02:41 sbfspot SBFspot[159557]: (c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Jun 10 17:02:41 sbfspot SBFspot[159557]: Compiled for Linux (LE) 64 bit with MySQL support
Jun 10 17:02:41 sbfspot SBFspot[159557]: Commandline Args: -v -ad0 -am0 -ae0 -mqtt -cfg:/etc/sbfspot/sbfspot.default.conf
Jun 10 17:02:41 sbfspot SBFspot[159557]: Reading config '/etc/sbfspot/sbfspot.default.conf'
Jun 10 17:02:41 sbfspot SBFspot[159557]: Sat Jun 10 17:02:41 2023: INFO: Starting...
Jun 10 17:02:52 sbfspot SBFspot[159557]: sunrise: 06:57
Jun 10 17:02:52 sbfspot SBFspot[159557]: sunset : 16:51
Jun 10 17:02:52 sbfspot SBFspot[159557]: Connecting to 00:80:25:0C:40:90 (1/10)
Jun 10 17:02:52 sbfspot SBFspot[159557]: Initialising...
Jun 10 17:02:52 sbfspot SBFspot[159557]: SUSyID: 125 - SessionID: 900222156
Jun 10 17:02:52 sbfspot SBFspot[159557]: SMA netID=02
Jun 10 17:02:52 sbfspot SBFspot[159557]: SUSyID: 99 - SN: <elided just in case>
Jun 10 17:02:52 sbfspot SBFspot[159557]: SUSyID: 99 - SN: <elided just in case>
Jun 10 17:02:52 sbfspot SBFspot[159557]: SUSyID: 99 - SN: <elided just in case>
Jun 10 17:02:52 sbfspot SBFspot[159557]: BT Signal=71.4%
Jun 10 17:02:52 sbfspot SBFspot[159557]: Logon OK
Jun 10 17:02:52 sbfspot SBFspot[159557]: Failed to get current plant time (-1)
Jun 10 17:02:52 sbfspot SBFspot[159557]: SetPlantTime returned an error: -1
Jun 10 17:03:30 sbfspot SBFspot[159557]: getSoftwareVersion returned an error: -1
Jun 10 17:03:59 sbfspot SBFspot[159557]: recv() returned an error: 110
Jun 10 17:03:59 sbfspot SBFspot[159557]: getTypeLabel returned an error: -1
Jun 10 17:03:59 sbfspot SBFspot[159557]: send() returned an error: 107

```

The SBFspot process then hangs, and doesn't exit (which breaks cronjob). So far my solution in systemd is to set a timeout of 120seconds to get the job to be killed when if gets into this state (I suspect the issue with multiple processes is people's cronjobs spawning new ones).

endrikus, 2023-07-21:
Could you explain your solution a bit @wrouesnel ?

It would be usefull until it is fixed.

MD1848, 2023-07-30:
my solution:
disable crontab jobs; setup systemd services:

cat /etc/systemd/system/sbfspot-daydata.service  

```
[Unit]
Description=sbfspot-daydata-service
###OnFailure=status_email_morten@%N

[Service]
Type=oneshot
User=sbfspot
Group=sbfspot
Restart=on-failure
RestartSec=30s
ExecStart=/usr/local/bin/sbfspot.3/daydata
SyslogIdentifier=SBFSPOT
# Override default 90 second timeout in pathological conditions
TimeoutStopSec=120

[Install]
WantedBy=multi-user.target
```

cat /etc/systemd/system/sbfspot-daydata.timer
```
[Unit]
Description=sbfspot-daydata-timer

[Timer]
OnBootSec=2min
###OnUnitActiveSec=5min
OnCalendar=5..22/5:00
Unit=sbfspot-daydata.service
AccuracySec=5s

[Install]
WantedBy=multi-user.target
```

wrouesnel, 2023-08-15:
Yeah my solution is similar:

```
# /etc/systemd/system/sbfspot-mqtt.timer
[Unit]
Description=SBFspot MQTT Upload Timer

[Timer]
AccuracySec=1s
OnCalendar=*:*:0/10

[Install]
WantedBy=timers.target
```

and 

```
# /etc/systemd/system/sbfspot-mqtt.service
[Unit]
Description=SBFspot MQTT Upload

[Service]
Type=oneshot
ExecStart=/opt/sbfspot/SBFspot -v -ad0 -am0 -ae0 -mqtt -cfg:/etc/sbfspot/sbfspot.default.conf
TimeoutStartSec=120s

User=sbfspot
Group=sbfspot
```

closed 2025-07-24 by sbf-