SBFspot/SBFspot #164: doesn't compile with gcc-c++ 6.1.1?

SBFspot, 2017-11-04:
after upgrading from fc23 to fc24, my sbfspot package doesn't compile.

g++ -Wall -O2 -DUSE_MYSQL  -c TagDefs.cpp -o obj/Release_MySQL/TagDefs.o
db_MySQL_Export.cpp: In member function 'int db_SQL_Export::day_data(InverterData**)':
db_MySQL_Export.cpp:95:28: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
   values[0].is_null  = false;
                    ^~~~~

db_MySQL_Export.cpp:102:28: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
   values[1].is_null  = false;
                    ^~~~~

db_MySQL_Export.cpp:109:28: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
   values[2].is_null  = false;
                    ^~~~~

db_MySQL_Export.cpp:116:28: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
   values[3].is_null  = false;
                    ^~~~~

db_MySQL_Export.cpp: In member function 'int db_SQL_Export::month_data(InverterData**)':
db_MySQL_Export.cpp:181:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
  values[0].is_null  = false;
                   ^~~~~

db_MySQL_Export.cpp:188:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
  values[1].is_null  = false;
                   ^~~~~

db_MySQL_Export.cpp:195:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
  values[2].is_null  = false;
                   ^~~~~

db_MySQL_Export.cpp:202:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
  values[3].is_null  = false;
                   ^~~~~

db_MySQL_Export.cpp: In member function 'int db_SQL_Export::event_data(InverterData**, TagDefs&)':
db_MySQL_Export.cpp:339:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[0].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:346:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[1].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:353:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[2].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:360:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[3].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:367:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[4].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:374:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[5].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:381:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[6].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:387:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[7].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:393:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[8].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:400:26: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[9].is_null  = false;
                  ^~~~~

db_MySQL_Export.cpp:407:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[10].is_null  = false;
                   ^~~~~

db_MySQL_Export.cpp:413:27: error: cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
 values[11].is_null  = false;
                   ^~~~~

cp TagList*.txt bin/Release_MySQL
cp date_time_zonespec.csv bin/Release_MySQL
makefile:221: recipe for target 'obj/Release_MySQL/db_MySQL_Export.o' failed
make: *** [obj/Release_MySQL/db_MySQL_Export.o] Error 1
make: *** Waiting for unfinished jobs....
SBFspot.cpp: In function 'E_SBFSPOT initialiseSMAConnection(InverterData*)':
SBFspot.cpp:1408:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
 if (DEBUG_NORMAL)
 ^~

SBFspot.cpp:1413:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
  do
  ^~
SBFspot.cpp: In function 'int GetConfig(Config*)':
SBFspot.cpp:2457:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
 if (strlen(cfg->plantname) == 0)
 ^~

SBFspot.cpp:2460:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if'
  if (cfg->timezone.empty())
  ^~
SBFspot.cpp: In function 'E_SBFSPOT setDeviceData(InverterData*, LriDef, uint16_t, Rec40S32&)':
SBFspot.cpp:3252:5: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
 else
 ^~~~

SBFspot.cpp:3255:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
  return rc;
  ^~~~~~
SBFspot.cpp: In function 'E_SBFSPOT getDeviceData(InverterData*, LriDef, uint16_t, Rec40S32&)':
SBFspot.cpp:3280:5: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
 else
 ^~~~

SBFspot.cpp:3283:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
  int validPcktID = 0;
  ^~~
error: Bad exit status from /var/tmp/rpm-tmp.BeSRrb (%build)

Is this related to the gcc-c++ version (6.1.1) on fc24?

#### This work item was migrated from CodePlex
CodePlex work item ID: '160'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@6/09/2016]
Yes, this is caused by gcc 6 (https://gcc.gnu.org/gcc-6/porting_to.html)
Compiler can be too strict IMO:
```
A new warning -Wmisleading-indentation was added to -Wall,
warning about places where the indentation of the code might mislead
a human reader about the control flow
```
Simplest solution might be adding
```
#pragma GCC diagnostic ignored "-Wmisleading-indentation"
```

For the other errors I'll need to investigate
```
cannot convert 'bool' to 'my_bool* {aka char*}' in assignment
```

SBFspot, 2017-11-04:
[svde@6/09/2016]
Following your suggestion I managed to get rid of the -Wmisleading-indentation errors.

The cannot conver 'bool' to 'my_bool' errors don't allow it to compile:

makefile:221: recipe for target 'obj/Release_MySQL/db_MySQL_Export.o' failed
make: *** [obj/Release_MySQL/db_MySQL_Export.o] Error 1
make: *** Waiting for unfinished jobs....

I'm seeing the same warnings when I build it on my pi1 (jessie).

SBFspot, 2017-11-04:
[SBF@6/09/2016]
Same on Jessie?
What version of gcc is installed?
MySQL version?

SBFspot, 2017-11-04:
[svde@6/09/2016]
reread my last post, it wasn't clear.

Jessie on rpi1 compiles, but throws warnings on the bool to my_bool.

FC24 doesn't compile, and breaks on the bool to my_bool errors.

SBFspot, 2017-11-04:
[SBF@9/09/2016]
I installed Fedora 24 with gcc 6.1 on VirtualBox and was able to reproduce the errors.
I had some troubles finding the right packages - still missing the libmysql library
I'll let you know when I have a fix.

SBFspot, 2017-11-04:
[svde@9/09/2016]
URL to the src.rpm I'm using: http://anorien.csc.warwick.ac.uk/mirrors/OBS/warwick.ac.uk:/CSC:/Public/CentOS_6/src/SBFspot-3.3.1-1.1.csc.src.rpm

SBFspot, 2017-11-04:
**Issue closed by _SBF_ with comment**
Fixed in V3.3.2

**Reason closed**
Fixed

closed 2017-11-04 by SBFspot

OldPaulie, 2018-02-20:
#164
Hello, I am suffering from the same compile issues as described above.  I have downloaded v3.3.1 but I would like to patch it to overcome the compile issue(s).  Are you able to help me?

sbf-, 2018-02-20:
Published 3.3.2 patch

OldPaulie, 2018-02-20:
That seemed to work!

Thank you

uname -a
Linux 4.8.0-2-amd64 #1 SMP Debian 4.8.15-2 (2017-01-04) x86_64 GNU/Linux
(Debian Stretch/ stable)

dpkg -l | grep gcc-6
ii  gcc-6
6.3.0-18                          amd64        GNU C compiler
ii  gcc-6-base:amd64
6.3.0-18                          amd64        GCC, the GNU Compiler
Collection (base package)
ii  libgcc-6-dev:amd64
6.3.0-18                          amd64        GCC support library
(development files)

~/sbfspot.3/SBFspot$ sudo make install_mysql
test -d bin/Release_MySQL || mkdir -p bin/Release_MySQL
test -d obj/Release_MySQL || mkdir -p obj/Release_MySQL
g++ -Wall -O2 -DUSE_MYSQL  -c db_MySQL_Export.cpp -o
obj/Release_MySQL/db_MySQL_Export.o
g++ -Wall -O2 -DUSE_MYSQL  -c misc.cpp -o obj/Release_MySQL/misc.o
g++ -Wall -O2 -DUSE_MYSQL  -c strptime.cpp -o obj/Release_MySQL/strptime.o
g++ -Wall -O2 -DUSE_MYSQL  -c sunrise_sunset.cpp -o
obj/Release_MySQL/sunrise_sunset.o
g++ -Wall -O2 -DUSE_MYSQL  -c SBFNet.cpp -o obj/Release_MySQL/SBFNet.o
g++ -Wall -O2 -DUSE_MYSQL  -c Bluetooth.cpp -o obj/Release_MySQL/Bluetooth.o
g++ -Wall -O2 -DUSE_MYSQL  -c CSVexport.cpp -o obj/Release_MySQL/CSVexport.o
g++ -Wall -O2 -DUSE_MYSQL  -c Ethernet.cpp -o obj/Release_MySQL/Ethernet.o
g++ -Wall -O2 -DUSE_MYSQL  -c EventData.cpp -o obj/Release_MySQL/EventData.o
g++ -Wall -O2 -DUSE_MYSQL  -c ArchData.cpp -o obj/Release_MySQL/ArchData.o
g++ -Wall -O2 -DUSE_MYSQL  -c SBFspot.cpp -o obj/Release_MySQL/SBFspot.o
g++ -Wall -O2 -DUSE_MYSQL  -c TagDefs.cpp -o obj/Release_MySQL/TagDefs.o
g++  -o bin/Release_MySQL/SBFspot obj/Release_MySQL/boost_ext.o
obj/Release_MySQL/db_MySQL.o obj/Release_MySQL/db_MySQL_Export.o
obj/Release_MySQL/misc.o obj/Release_MySQL/strptime.o
obj/Release_MySQL/sunrise_sunset.o obj/Release_MySQL/SBFNet.o
obj/Release_MySQL/Bluetooth.o obj/Release_MySQL/CSVexport.o
obj/Release_MySQL/Ethernet.o obj/Release_MySQL/EventData.o
obj/Release_MySQL/ArchData.o obj/Release_MySQL/SBFspot.o
obj/Release_MySQL/TagDefs.o   -s -lbluetooth -lboost_date_time
-lboost_system -lpthread -lmysqlclient
cp TagList*.txt bin/Release_MySQL
cp date_time_zonespec.csv bin/Release_MySQL
test -d bin/Release_MySQL || release_mysql
test -d /usr/local/bin/sbfspot.3 || mkdir -p /usr/local/bin/sbfspot.3
test -f /usr/local/bin/sbfspot.3/SBFspot.cfg || cp SBFspot.cfg
/usr/local/bin/sbfspot.3/
cp -R bin/Release_MySQL/* /usr/local/bin/sbfspot.3/