SBFspot/SBFspot #169: Uploading during the night.

SBFspot, 2017-11-04:
Is it possible to upload data during the night?
currently the uploads stop at a yield of 0 watts.

I would like to extend the DataBase with temperature, consumption, humidity, light intensity, wind speed and other data.

I want to see this data from the whole night.

just using -v -finq does not seem to help, spotdata is filled but the upload deamon is not uploading.

#### This work item was migrated from CodePlex
CodePlex work item ID: '165'
Vote count: '3'

SBFspot, 2017-11-04:
[SBF@18/10/2016]
With the current version it's not possible.
You will need some code changes in order to store the missing data

SBFspot, 2017-11-04:
[SteloNLD@21/10/2016]
Thank you for your reply,

I took a short peek at the source code and I am thinking about forking en rewriting the code for my usage case.

keep up the good work!

SBFspot, 2017-11-04:
[bobonia@7/02/2017]
Hi SteloNLD, any progress so far?

I installed SBFspot on an emonpi raspberry-pi based energy monitor and it works great.
I'm currently uploading production values to pvoutput and would really love to use SBFspot to also upload consumption data read by the emonpi.
I noticed there are already avg energy and power columns in the vwPvoData view which gets uploaded to pvoutput, so I put a bash script togheter that reads the consumption values from my energy monitor and stores them in the Consumption table with a direct sql query.

With no other changes, when the uploader runs my consumption data gets uploaded to pvoutput, along with the data from the inverter, but... only while there is an actual power output from the inverter.
Data stored in the DB during the night by my script does not get uploaded to pvouput, not even the following morning.

I would love to achieve continuous uploading of any new row in the vwPvoData view (without unneccessary polling of the inverter during the night), and I am willing to contribute code changes if anyone could just point me in the right direction to start with.

Thanks a lot and keep up the excellent work!

mr-russ, 2019-09-27:
I'm currently looking into the options here.  Whether an upload is needed it stored in dayData and joined for vwPvoData. 
I would be really interested to know initial results if you add a record with time and pvoutput 0 into that table and see if it uploads.  My reading of the code says it will.

If you can post results that woyld be great.

closed 2021-04-18 by sbf-