SBFspot/SBFspot #173: fixed path "date_time_zonespec.csv" ?

SBFspot, 2017-11-04:
Confirmed as bug on Windows/Linux - All SBFspot versions
See this thread

#### This work item was migrated from CodePlex
CodePlex work item ID: '169'
Vote count: '1'

SBFspot, 2017-11-04:
[neumi@29/01/2017]
Hello,

Is it necessary to keep a separate file only for TimeZone information and look for path where to find it?
I think user needs only one line....

So why not just add it to *.config file as comments and user just uncomments his choice?

One file less to care about....

SBFspot, 2017-11-04:
[SBF@29/01/2017]
_date_time_zonespec.csv_ is part of [BOOST](http://www.boost.org)
Some functions like _time_zone_from_region()_ rely on this file.

SBFspot, 2017-11-04:
[neumi@6/02/2017]
I see....
Why not put some keyword into SBFspot.cfg...
```
# TimezoneDB
# Select  path and name of BOOST's "date_time_zonespec.csv" file
TimezoneDB=/user/locale/etc/date_time_zonespec.csv
```
So, independent of binary file the tzDB could be placed e.g. in same folder as SBFspot.cfg....

Additionally in source of "SBFspot.cpp" I think the error message 
```
...
                     {
                            cout << "Unable to open 'date_time_zonespec.csv'\n";
                            return -2;
                    }
...
```
should be sent to "stderr" instead of "stdout" (I am not familiar with c++ but "cout" seems to use "stdout").... especially when using "-q" option and 123solar.

sbf-, 2018-02-25:
Fixed in V3.4.0

closed 2018-02-25 by sbf-