SBFspot/SBFspot #174: Not completely silent with -q

SBFspot, 2017-11-04:
Hello,

Is this a bug or a feature that following error messages are always written to "stdout" instead of being suppressed ("-q") or send to "stderr" so it can be treated separately, e.g. in connection with 123solar....
SetPlantTime returned an error: -1
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1

Following scenario to show what I means:

situation 1: everything as expected
pi@raspberrypi2:~ $ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -nosql
SBFspot V3.3.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2016, SBF (https://sbfspot.codeplex.com)
Compiled for Linux (LE) 32 bit

Commandline Args: -v -finq -nocsv -nosql
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Sun Jan 29 18:28:34 2017: INFO: Starting...
sunrise: 07:54
sunset : 17:10
Connecting to 00:80:25:0A:1A:C5 (1/15)
Initializing...
SUSyID: 125 - SessionID: 838052592 (0x31F3AAF0)
SMA netID=01
Serial Nr: 7D2D036C (2100101996)
BT Signal=69.8%
Logon OK
Failed to get current plant time (0)
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1
SUSyID: 78 - SN: 2100101996
Device Name:      SN: 2100101996
Device Class:     Solar Inverters
Device Type:      SB 4000TL-20
Software Version:
Serial number:    2100101996
SUSyID: 78 - SN: 2100101996
Device Status:      Ok
SUSyID: 78 - SN: 2100101996
Device Temperature: 0.0Â°C
SUSyID: 78 - SN: 2100101996
GridRelay Status:      Information not available
SUSyID: 78 - SN: 2100101996
Pac max phase 1: 4000W
Pac max phase 2: 4000W
Pac max phase 3: 4000W
SUSyID: 78 - SN: 2100101996
Energy Production:
    EToday: 14.905kWh
    ETotal: 30365.740kWh
    Operation Time: 29339.49h
    Feed-In Time  : 28291.31h

SUSyID: 78 - SN: 2100101996
DC Spot Data:
    String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
    String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A

SUSyID: 78 - SN: 2100101996
AC Spot Data:
    Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
    Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
    Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
    Total Pac   :   0.000kW

SUSyID: 78 - SN: 2100101996
Grid Freq. : 0.00Hz
SUSyID: 78 - SN: 2100101996
Current Inverter Time: 29/01/2017 19:05:31
Inverter Wake-Up Time: 29/01/2017 07:56:37
Inverter Sleep Time  : 29/01/2017 17:14:37

ArchiveDayData() *

startTime = 588D2270 -> 29/01/2017 00:00:00

ArchiveMonthData() *

startTime = 5868E130 -> 01/01/2017 12:00:00
Reading events: 2017-Jan-01
Sun Jan 29 18:29:38 2017: INFO: Done.

situation 2: output in quiet mode is OK?!?
(test is send to stdout since it cannot be redirected with a " 2>err.txt "
pi@raspberrypi2:~ $ /usr/local/bin/sbfspot.3/SBFspot -q -finq -nocsv -nosql
SetPlantTime returned an error: -1
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1

situation 3: when being used with 123solar.... 
pi@raspberrypi2:~ $ /usr/local/bin/sbfspot.3/SBFspot -q -finq -nocsv -nosql -123s=DATA
SetPlantTime returned an error: -1
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1
20170129-19:07:06 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 30365.740 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 >>>S123:OK

Question:
Is this behavior OK ? 

#### This work item was migrated from CodePlex
CodePlex work item ID: '170'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@29/01/2017]
See also https://sbfspot.codeplex.com/discussions/661017

sbf-, 2018-02-25:
Fixed in V3.4.0

closed 2018-02-25 by sbf-