SBFspot/SBFspot #175: SBFspot doesn't create spot file for Sunny Multigate

SBFspot, 2017-11-04:
SBFspot 3.2.0 doesn't create spot file for Sunny Multigate. 
line 531 in SBFspot.cpp 
if (Inverters[0]->DevClass == SolarInverter)

Multigate is a CommunicationProduct Device.

I changed it to 
if (Inverters[0]->DevClass == SolarInverter || Inverters[0]->DevClass == CommunicationProduct )

and now is working fine. I'm not sure if this change is safe for others. I have only Multigate so at least for me it is fine

#### This work item was migrated from CodePlex
CodePlex work item ID: '171'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@31/01/2017]
Don't know if this is a good idea... you will have double Energy Today/Total (for each micro inverter and the total of all inverters as reported by your Multigate)
Also, Mutigate devices don't report Power/Voltage and Current for AC/DC if I'm right?

If you want to change this, I think its better to check for Multigate SUSyID (175)
```C++
if (Inverters[0]->DevClass == SolarInverter || Inverters[0]->SUSyID == 175)
```
This way you avoid other "CommunicationProducts"

SBFspot, 2017-11-04:
[amakowsk@31/01/2017]
Thank you for quick response. Truly speaking I don't need spot file. I'm reading direct from DB so I can take whatever I  want. But I tested Domoticz.com solution and they have build in reading for SMA via SBFspot. The only issue is that they read from spot file. So basically it doesn't work on 3.2.0 build. But after changes in code and using "SBFspot.cfg:1915015483" in their configuration it start to work. Serial number is to avoid double energy. It will be ready only for multigate with that serial. 

Multigate reports AC and DC side however I'm not sure what is on DC side as it is rather virtual and probably avg of DCs on inverters

On AC side:
SUSyID: 175 - SN: 1915015483
AC Spot Data:
        Phase 1 Pac :   1.071kW - Uac: 224.46V - Iac:  4.753A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   1.068kW

SUSyID: 175 - SN: 1915015483
DC Spot Data:
        String 1 Pdc:   0.000kW - Udc:  32.07V - Idc:  0.000A
        String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A

Anyway I want thank you for great job with software. I can monitor my energy plant much better than using  sunnyportal.

sbf-, 2018-07-22:
For now, I will leave it as is.
If this is still needed by someone, feel free to reopen.

closed 2018-07-22 by sbf-