SBFspot/SBFspot #177: MySQL EToday and ETotal as integer

SBFspot, 2017-11-04:
I just find strange behaviour. I'm not an expert in DB engines but it looks wired. 

Inverters table has the same definition for Sqlite and MySQL.
CREATE Table Inverters (
Serial int(4) NOT NULL,
Name varchar(32),
Type varchar(32),
SW_Version varchar(32),
TimeStamp datetime,
TotalPac int,
EToday int(8),
ETotal int(8),
OperatingTime double,
FeedInTime double,
Status varchar(10),
GridRelay varchar(10),
Temperature float,
PRIMARY KEY (Serial)

);

However for Sqlite in my DB I have for example this values for EToday and ETotal
1.885|50.353
0.177|3.732
0.172|6.233 
What is quite strange as it should be int(8). Issue is that MySQL keep this as real integer like below
1|50
0|3
0|6

I had to change data type to float and now MySql store data properly. 
In SpotData all is fine because there are Wh and can be kept as int but in Inverters we have kWh.

Anyway there is Segmentation fault in build for mysql v3.2 fro sure after data upload to DB. 

#### This work item was migrated from CodePlex
CodePlex work item ID: '173'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@2/02/2017]
Oh yes! This is a bug since the very first 3.0.0 - Thanks!
It's OK in SQLite because it uses a dynamic type system. You can store almost every type in any field.
See [Datatypes in SQLite 3](https://www.sqlite.org/datatype3.html)

Now I have 2 options:
* Change the DB structure to save EToday and ETotal in kWh
* Change the device_status() code to save EToday and ETotal in Wh

I tend to choose the second one for ease, but I'm open to suggestions.

About segfault: do you still have it after the changes?

SBFspot, 2017-11-04:
[amakowsk@2/02/2017]
From my point of view change from int to float for mysql is the easiest way. You don't need change anything else. Of course if You don't use these values in some export functions but I guess that You use SpotData table. One argument for this is it won't impact any current solutions based on reading current status from Inverters table. Like my :) If you will change to Wh I will have jump in charts from the time of new SBFspot compilation. From other hand second option will unified a way how values are stored. 

Reagrding segfault: Yes, I have
/usr/local/bin/sbfspot.3/SBFspot -v -nocsv -finq

At this moment 
********************
* ArchiveDayData() *
********************
startTime = 58926870 -> 02/02/2017 00:00:00
Segmentation fault

Witch csv
/usr/local/bin/sbfspot.3/SBFspot -v -finq
At this moment
********************
* ArchiveDayData() *
********************
startTime = 58926870 -> 02/02/2017 00:00:00
ExportDayDataToCSV()
Segmentation fault

SBFspot, 2017-11-04:
[amakowsk@2/02/2017]
Another info about segfault but I really don't understand this. I'm login via ssh as normal user then su to root
/usr/local/bin/sbfspot.3/SBFspot -v -finq gives error but when I do sudo /usr/local/bin/sbfspot.3/SBFspot -v -finq the there is no error. Whoami gives the same result "root" ...

SBFspot, 2017-11-04:
[SBF@3/02/2017]
Why do you run SBFspot as root? It's not recommended (security reasons)

SBFspot, 2017-11-04:
[amakowsk@3/02/2017]
Because ... emm because, I have powerbroker here and all root activities have to be reviewed by independent log review team (my wife) and compare with change management tickets. That have to be approved by independent change management team (my daughter) before login to Production environment. That is why I can use root. 

OK that was a joke :)

Seriously, security is not a reason. Security is a solution after risk analysis. 
I did my risk analysis, so on my Virtual sandbox I play with all tools that I want to test on one user. 
Btw Pi raspberry have default user with sudo allowed without any password. What is the same situation. 
Anyway I checked how SBFspot works on normal user and indeed works better... what is quite strange.

So I promise that when I will find final solution I will use SBFspot on standard non-root user. But I still can't decide where to put Synology NAS, OpenWrt router, or Raspbbery PI... 

And as we have so friendly and philosophic discussion  (to be root or not to be) Did You consider a much simpler tool that can be used as command line tool? I know that today I can switch off sql and csv export and just parse output from program. But it could be nice to have tool that just ask one inverter for current status. Currently it takes 7.2 sec to finish SBFspot on my Multigate.

SBFspot, 2017-11-04:
[SBF@3/02/2017]
Ok, no problem, I guess you know what you're doing ;-)
Don't know why it segfaults when running as 'su root' - have to check this

To decide where to run depends on what you want to do... OpenWrt router with SQLite db with upload to PVoutput is a nogo for me. Synology NAS is OK if you are able to compile (few people did it, but until now undocumented - so feel free to try it and document it)

A running time of 7.2 sec seems to be more than long enough for ethernet connected devices, but without knowing the installation it's difficult to judge. Multigate with x micro inverters?
The initial goal of SBFspot was to get as much as possible actual data out of the inverter. That takes time.

SBFspot, 2017-11-04:
[amakowsk@4/02/2017]
Backing to our root cause of issue. DON'T change definition from int to float. And consider change all float definition to decimal (x,y). It is not so important for Sqlite3 because it maps float to real for storage. And "A column with REAL affinity behaves like a column with NUMERIC affinity " - https://www.sqlite.org/datatype3.html

The real problem is in MySQL - 
CREATE  TABLE `test_table` (`number` FLOAT NULL);
Insert into test_table value (0.7);
select number *1000 from test_table;
'699.999988079071'
select number + number from test_table
'1.399999976158142'

Regarding installation I have 12 inverters (full capacity of multigate)

sbf-, 2018-02-25:
Fixed in V3.4.0

closed 2018-02-25 by sbf-