SBFspot/SBFspot #178: MySQL day_data is not working with Multigate

SBFspot, 2017-11-04:
Hi, 
Environment: 
SMA Multigate + 12 inverters
MySQL 5.5 on Synology NAS
SBFspot v 3.2 on Raspberry Pi

When I started above configuration all was fine. But after few cycles stopped to work. I checked logs and found that MySQL started refused connection because "aborted_clients" was set to 10. It means that after 10 connection that wasn't closed properly host who did that is added to blocked hosts. I changed to 1000 and started look what is wrong with closing connection is SBFspot. 
It looks that program stop to work between 
line 573 db.spot_data(Inverters, spottime);
line 736 db.close();

I guess that is somewhere in db.day_data as all day data for inverters are printed correctly when verbose is high. But here I stopped as I'm not sure what is going on in this function. I always had a empty table for Day_data (month data too) and thought that is a normal. 
There is also other thing when you start program as normal user there is segmentation fault at this moment but when use sudo there is no. Anyway program is not finishing normally and ther is nor release of db connection.
Is there a way how I could help to find a root cause? I attached debug 5 and verbose 5

Regards

#### Attachments
[debug.txt](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=174&FileAttachmentId=1645667)

#### This work item was migrated from CodePlex
CodePlex work item ID: '174'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@13/03/2017]
Is the aborted_clients issue caused by segfault? When segfault occurs between opening/closing db?

SBFspot, 2017-11-04:
[SBF@13/03/2017]
Please contact me via PM
I have an updated beta version 3.2.1

sbf-, 2018-02-25:
Fixed in V3.2.1

closed 2018-02-25 by sbf-