SBFspot/SBFspot #179: Export to Time Series DBs like graphite

SBFspot, 2017-11-04:
Collecting such performance data it would be nice to get an export option to time-series DBs like graphite.
Those can be consumed by grafana to visualize the data.

#### Attachments
[Screen Shot 2017-03-14 at 15.35.34.png](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=175&FileAttachmentId=1646184)

#### This work item was migrated from CodePlex
CodePlex work item ID: '175'
Vote count: '1'

SBFspot, 2017-11-04:
[mcsage@14/03/2017]
Currently I use a wrapper script which uses nc (netcat). This might be sufficent to integrate.

http://graphite.readthedocs.io/en/latest/feeding-carbon.html#the-plaintext-protocol

ThetaGamma, 2017-11-09:
Sounds good - could you provide the wrapper script? One more thought about this: Outputting JSON might be a solution, too

sbf-, 2017-11-09:
Most of the issues are imported from codeplex by SBFspot user. Try to contact mcsage via [codeplex](https://sbfspot.codeplex.com/workitem/175)

ThetaGamma, 2017-11-09:
OK, got it!
It would also help to get the csv-output directly to commandline, not written to the file

closed 2018-02-25 by sbf-