SBFspot/SBFspot #183: not identying new SMA SB 3.6-1 40-AV

SBFspot, 2017-11-04:
SBFSpot currently is not identifying the new SMA inverters e.g. SB3.6-1AV-40

Part of the SBFSpot-output:

Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 984734624 (0x3AB1DBA0)
Inverter IP address: 192.168.2.235 from SBFspot.cfg
Logon OK
Unknown Inverter Type. Report this issue at https://sbfspot.codeplex.com/workitem/list/basic with following info:
0x00002468 and Inverter Type=<Fill in the exact type> (e.g. SB1300TL-10)
SUSyID: 342 - SN: 1992014865
Device Name:      SB3.6-1AV-40 865
Device Class:     Solar-Wechselrichter
Device Type:      UNKNOWN TYPE
Software Version: 01.03.03.R
Serial number:    1992014865
SUSyID: 342 - SN: 1992014865
Device Status:      Ok
SUSyID: 342 - SN: 1992014865

#### Attachments
[TagList_340.zip](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=179&FileAttachmentId=1670149)

#### This work item was migrated from CodePlex
CodePlex work item ID: '179'
Vote count: '1'

SBFspot, 2017-11-04:
[Brutuss@11/07/2017]
Same problem with the SB5.0-1AV-40

SBFspot, 2017-11-04:
[SBF@11/07/2017]
Taglists have been updated with new devices. Extract [TagList_340.zip](https://sbfspot.codeplex.com/releases) and replace the existing ones in /usr/local/bin/sbfspot.3 (Raspberry Pi) or any other location of SBFspot executable

SBFspot, 2017-11-04:
**Issue closed by _SBF_ with comment**
Added new taglists to 3.3.1 release

**Reason closed**
Fixed

closed 2017-11-04 by SBFspot

sbf-, 2018-02-25:
Fixed in V3.4.0