SBFspot/SBFspot #184: Unknown Inverter Type

SBFspot, 2017-11-04:
SBFspot Doesn't know the new Sunny Boy inverters: 

I have the New SMA Sunny Boy 5.0 (SMA SB 5.0-1AV-40)

C:\Users\Domoticz\Documents\SBFspot>sbfspot
SBFspot V3.3.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2016, SBF (https://sbfspot.codeplex.com)
Compiled for Windows (LE) 32 bit

Commandline Args:
Inverter IP address: 192.168.1.3 found via broadcastidentification
Unknown Inverter Type. Report this issue at https://sbfspot.codeplex.com/workite
m/list/basic with following info:
0x0000246A and Inverter Type=<Fill in the exact type> (e.g. SB1300TL-10)

#### This work item was migrated from CodePlex
CodePlex work item ID: '180'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@17/07/2017]
See issue 179 for fix

SBFspot, 2017-11-04:
**Issue closed by _SBF_ with comment**
See issue 179 for fix

**Reason closed**
Fixed

closed 2017-11-04 by SBFspot

sbf-, 2018-02-25:
Fixed in V3.4.0