SBFspot/SBFspot #185: Tool goes to broadcast and search inverter when 10.1.1.1-9 IPs used

SBFspot, 2017-11-04:
Hi,

  Spent quite some time to figure out a bug in the code : 
if (strlen(IP_Address) > 8) // len bigger than 0.0.0.0 or len of no string ==> use IP_Adress from config
{
    sprintf(inverters[devcount]->IPAddress, "%s", IP_Address);
    if (quiet == 0) printf("Inverter IP address: %s from SBFspot.cfg\n", inverters[devcount]->IPAddress);
}
        else    // use IP from broadcast-detection of inverter
{
        sprintf(inverters[devcount]->IPAddress, "%d.%d.%d.%d", CommBuf[38], CommBuf[39], CommBuf[40], CommBuf[41]);
    if (quiet == 0) printf("Inverter IP address: %s found via broadcastidentification\n", inverters[devcount]->IPAddress);
}

if (strlen(IP_Address) > 8) then if should use IP from config. But what if ip is 10.1.1.8 ? ;-) I guess it should be '>= 8 ' in the code . Or '>7'

br,
dmitry

#### This work item was migrated from CodePlex
CodePlex work item ID: '181'
Vote count: '1'

SBFspot, 2017-11-04:
[SBF@30/07/2017]
Thanks for reporting this bug.
Maybe better just use strcmp()

sbf-, 2018-02-25:
Fixed in V3.4.0

closed 2018-02-25 by sbf-