SBFspot/SBFspot #43: Line endings in UNIX are not CR/LF

SBFspot, 2017-11-04:
Hi,

When running SMAspot in linux the resulting lines in the CSV files are not ended with CR/LF but with LF only like is custom with linux text files. 

The header is indicating that the line endings are CR/LF but they are not. I don't see any flag in the config to force alternate line endings. 

Cheers,

Wim

#### Attachments
[2014-05-27_1153.png](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=37&FileAttachmentId=850367)

#### This work item was migrated from CodePlex
CodePlex work item ID: '37'
Vote count: '1'

SBFspot, 2017-11-04:
[wimjongman@27/05/2014]
Hi I did some research and the reason is that \n is CR/LF on windows and LF on unix. 

I'm no C++ guy but if you don't have time to fix this issue please say so and I will take a stab at it if you are willing to accept my patch. Is there something like a github pull request in codeplex?

Cheers,

wim

SBFspot, 2017-11-04:
[SBF@27/05/2014]
Hi Wim,
thanks for reporting this issue.
As I don't see this as blocking problem, I can only handle this with low priority.
If you are able to fix it, go ahead. I'm using Mercurial as source control. See "Source code" section
regards,
sbf

sbf-, 2018-02-25:
Fixed in V3.4.0

closed 2018-02-25 by sbf-