SBFspot/SBFspot #7: MariaDB10 with Port 3307

grueger-pv, 2017-11-04:
Hello.
to use MariaDB10, the port is changed to 3307. Can I change the port in SBFspot, 3.12 is running here. Adding the port in my.cnf don't work.

greets, mike

sbf-, 2017-11-04:
SBFspot has no options to configure the port. Only hostname and default settings

bartd83, 2017-11-06:
I probably shouldn't be asking my question in this issue, I thought the compilation was only possible with mysql and sqlite ? Not MariaDB ? At least I can only see install_mysql and install_sqlite.

grueger-pv, 2017-11-07:
MariaDB based on MySQL. Install_mysql works for MaroaDB 5.5. For MariaDB 10.x db_MySQL.cpp must be edit to change the Port.

bartd83, 2017-11-07:
hmm, then I need to look into it some further, cause it failed during compile saying several MySQL folders didn't exist.

grueger-pv, 2017-11-07:
libmysqlclient-dev is installed? Maybe mysql-client too.

bartd83, 2017-11-14:
yeah, overlooked through installation -> it's libmariaclient-dev which is needed
Compilation still fails however : 
********************************
g++ -Wall -O2 -DUSE_MYSQL  -c boost_ext.cpp -o obj/Release_MySQL/boost_ext.o
g++ -Wall -O2 -DUSE_MYSQL  -c db_MySQL.cpp -o obj/Release_MySQL/db_MySQL.o
g++ -Wall -O2 -DUSE_MYSQL  -c db_MySQL_Export.cpp -o obj/Release_MySQL/db_MySQL_                                                                                                                               Export.o
db_MySQL_Export.cpp: In member function âint db_SQL_Export::day_data(InverterDat                                                                                                                               a**)â:
db_MySQL_Export.cpp:95:28: error: cannot convert âboolâ to âmy_bool* {aka char*                                                                                                                                â in assignment
       values[0].is_null  = false;
                            ^~~~~
db_MySQL_Export.cpp:102:28: error: cannot convert âboolâ to âmy_bool* {aka char*                                                                                                                               }â in assignment
       values[1].is_null  = false;
***************************
think I'm just gonna replace those mariadb packages to get back to mysql

commit f4301f6a (2021-01-17): Fix #7 MariaDB10 with Port 3307
closed 2021-01-17 by sbf-