SBFspot/SBFspot #98: Add PID-File for SBFspotUploadDaemon

SBFspot, 2017-11-04:
May you please add an option to create a PID-File for the SBFspotUploadDaemon.
I've done this actually by a wrapper script, but this sucks a little bit during upgrade process.

/usr/local/bin/sbfspot.3/SBFspotUploadDaemon.sh
!/bin/bash
start upload daemon
/usr/local/bin/sbfspot.3/SBFspotUploadDaemon
write pid file
PID=pidof SBFspotUploadDaemon

echo "$PID" > /run/SBFspotUploadDaemon.pid

Maybe you can add this Init-Script for Debian based systems (also Raspberry PI). This took care for the PID-File and can start and stop the SBFspotUploadDaemon in a correct way. No more need for patching the /etc/rc.local.
!/bin/sh
BEGIN INIT INFO
Provides:          SBFspotUploadDaemon
Required-Start:
Required-Stop:
Should-Start:      $all
Should-Stop:       $all
Default-Start:     2 3 4 5
Default-Stop:      0 1 6
Short-Description: upload gathered data from SMA inverters
Description:

END INIT INFO
set -e

. /lib/lsb/init-functions

DAEMON=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon.sh
NAME=SBFspotUploadDaemon
DESC="daemon upload"
PID="/run/$NAME.pid"
SBFUD_OPTS=""
Check if DAEMON binary exist
[ -f $DAEMON ] || exit 0

[ -f "/etc/default/$NAME" ] && . /etc/default/$NAME

case "$1" in
  start)
log_daemon_msg "Starting $DESC" "$NAME"
if start-stop-daemon --start --quiet --oknodo --pidfile $PID --exec $DAEMON -- $SBFUD_OPTS 1>/dev/null
then
  log_end_msg 0
else
  log_end_msg 1
fi
;;

stop)
log_daemon_msg "Stopping $DESC" "$NAME"
if start-stop-daemon --retry TERM/5/KILL/5 --oknodo --stop --quiet --pidfile $PID 1>/dev/null
then
  log_end_msg 0
else
  log_end_msg 1
fi
;;

restart|force-reload)
log_daemon_msg "Restarting $DESC" "$NAME"
start-stop-daemon --retry TERM/5/KILL/5 --oknodo --stop --quiet --pidfile $PID 1>/dev/null
if start-stop-daemon --start --quiet --oknodo --pidfile $PID --exec $DAEMON -- $SBFUD_OPTS 1>/dev/null
then
  log_end_msg 0
else
  log_end_msg 1
fi
;;

status)
status_of_proc -p $PID $DAEMON $NAME
;;

*)
log_action_msg "Usage: /etc/init.d/$NAME {start|stop|restart|force-reload|status}"
;;

esac

exit 0

Additional benefit, process is now monitorable by Monit with the following modification done to /etc/monit/monitrc
...
check process SBFspotUploadDaemon with pidfile /run/SBFspotUploadDaemon.pid
start program = "/etc/init.d/SBFspotUploadDaemon start"
stop program = "/etc/init.d/SBFspotUploadDaemon stop"
...

Think this would be helpful to improve this great software.

#### Attachments
[sbfspot-pidfile.diff](https://www.codeplex.com/Download/AttachmentDownload.ashx?ProjectName=SBFspot&WorkItemId=93&FileAttachmentId=1382470)

#### This work item was migrated from CodePlex
CodePlex work item ID: '93'
Vote count: '4'

SBFspot, 2017-11-04:
[SBF@5/01/2015]
Very nice idea - didn't know about this being a Linux noob

SBFspot, 2017-11-04:
[wpaesen@5/01/2015]
Attached patch provides the functionality.  

It adds commandline parsing using getopt.  Two arguments are recognized :

-c, --config-file : where to find the config file, useful for putting the config file somewhere in /etc/ where it belongs.  When this option is not specified the daemon uses it's default behavior.

-p, --pid-file : where to write the pid file, that allows the daemon to integrate with startup scripts.  When this option is not specified no pid file is written.

SBFspot, 2017-11-04:
[jschanz@6/01/2015]
@wpaesen thank you very much for the patch. I think this will be very helpful. I'll try to wirte a blog post in the next few days about the solution. This could than be adapted to the official sbfspot documentation.

Any needs to "fix" the file structure to match the common linux structure?

e.g. putting the binary to /usr/bin, the configuration to /etc/sbfspot/ and the logfile(s) to /var/log/sbfspot/

SBFspot, 2017-11-04:
[SBF@6/01/2015]
Indeed a nice improvement, however, if --pid-file argument is not supplied, I suggest a default PID should be created:
```
/run/SBFspotUploadDaemon.pid
```

```
Any needs to "fix" the file structure to match the common linux structure?
```
Is there a common linux structure? I didn't find one while searching the web... At least not among different distro's ;-)
So, let's keep us like it is to not confuse existing users. It will complicate things when upgrading.

SBFspot, 2017-11-04:
[wpaesen@6/01/2015]
That's exactly the reason why no pidfile is created when no arguments are specified.  No arguments makes it behave as before this patch, so existing users can upgrade to the latest version without disturbing their installation.

As for structure, there are no binding rules, only de-facto standards imho.
Most build scripts would have an install target in their makefile that installs the binaries in the filesystem tree.   For example programs in /usr/bin, data like the taglists in /usr/share/sbfspot/, the sqlite database in /var/lib/sbfspot/ and the configuration file in /etc/.

We could cook something with autotools that sits next to the current structure that implements default 'nix package behavioure.

SBFspot, 2017-11-04:
[jschanz@6/01/2015]
You're right @SBF ... There is no common structure over all Linux distributions :-) But like @wpaesen suggested there are some common paths over all Linux distributions.
Have you some statistics about the most common platform of this project? Does most people run this on Raspbian(Debian) on a Raspberry Pi or on any other platform?
If @wpaesen could do his changes to the project, you could offer a pre-built Debian-Package for the Raspberry Pi oder any other needed platform for and maybe a Debian-Repository to solve the manual upgrade process. So no compiling is necessary for all the people.
If you want to do this and you need help, please tell me.

SBFspot, 2017-11-04:
[SBF@7/01/2015]
@jschanz
No statistics, but I'm sure Raspberry Pi is on top.
I did already some research to provide an easy installation and update system (_sudo apt-get install sbfspot_ would be the MAX) but it seems not to be that easy....
So if you're familiar with this and you have some spare time I greatly appreciate your help.

SBFspot, 2017-11-04:
[wpaesen@7/01/2015]
I'm currently trying to put autotools scripts in place in the source tree.    That would handle "default linux" file placement logic.    After that building debian packages is not too difficult anymore.  Takes some time though.

SBFspot, 2017-11-04:
[jschanz@9/01/2015]
Great ... Checked cross compiling for the arm achitecture. Works like a charm in Virtualbox. I'll try to make a complete vagrant box environment in the next few days. So automatic pre-build packages for the raspberry could be easily possible.
If we could combine this with the autotools, we can create a simple automatic build process for the raspberry environment.

SBFspot, 2017-11-04:
[haggan@6/09/2016]
So 

what is the status of this patch? 

I also want to provide a simple systemd service since that is the new Ubuntu standard
filename  /etc/systemd/system/SBFuploadDaemon.service
```
[Unit]
Description=SBFSpot Upload to pvoutput service
After=syslog.target

[Service]
Type=forking
PIDFile=/var/run/SBFspotUploadDaemon.pid
ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon.sh
Restart=on-abort

[Install]
WantedBy=multi-user.target

```

SBFspot, 2017-11-04:
[neumi@28/01/2017]
Hello,

Is the PID-patch meanwhile included in the official v3.3.1 package or do I need to add by myself...?

SBFspot, 2017-11-04:
[SBF@28/01/2017]
No, not yet... seems to be forgotten.
I'll add it to the upcoming 3.4

Abrastreve, 2019-08-20:
I know this issue is closed but when I try to monitor the Upload Daemon using Monit I do get a message saying: Program does not exist: '/etc/init.d/SBFspotUploadDaemon' after adding the following three lines to the monitrc file, as suggested in the first post of this issue:

check process SBFspotUploadDaemon with pidfile /run/SBFspotUploadDaemon.pid
start program = "/etc/init.d/SBFspotUploadDaemon start"
stop program = "/etc/init.d/SBFspotUploadDaemon stop"

I am using V3.5.2 of SBFspot
What am I doing wrong?

sbf-, 2019-08-21:
Upload daemon V2.0 doesn't use PID file anymore.
See [documentation](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)