SBFspot/SBFspot #187: Upload Daemon suddenly stopped uploading

davejcurry, 2017-11-16:
About a day ago, my data stopped uploading to PVOutput.

The logs only contain this message:

addBatchStatus() returned Timeout was reached

Is there a way to get more verbose logging, or any idea why it would suddenly stop? I've checked my API key and login details on pvoutput.org and they're all correct.

sbf-, 2017-11-16:
This is a PVoutput error. It should recover automatically.
Is the internet connection OK?

davejcurry, 2017-11-16:
Internet connection is fine. All my consumption data is uploading (I have an efergy hub for monitoring consumption).

davejcurry, 2017-11-16:
You are correct, the data is uploaded from my NAS, which is connected to the internet via a VPN. The VPN connection was down. Closing this, not an issue.

closed 2017-11-16 by davejcurry