SBFspot/SBFspot #188: Log grid frequency?

npkamen, 2017-11-24:
Hi there,

Has anyone brought up logging grid frequency and uploading it to PVOutput?  I just started running SBFSpot on my off grid system and it would be very handy to log grid frequency as this is what the off grid system uses to throttle my grid connect inverter when the batteries are getting full.

My off grid inverter doesn't have a comms interface so I can't log its data unfortunately but if I can get this frequency information (which does get reported when I run SBFspot manually) then I can tell if a drop in output is due to shading/clouds or frequency.

I'm happy to do the mod myself if required but just thought I'd ask if this has been brought up before and whether it was being considered for an upcoming release.

sbf-, 2017-11-24:
Isn't it in spotdata csv or sql table?

npkamen, 2017-11-24:
Ah yes I see it is.  How can I get sbfspot to upload this to PVOutput?

sbf-, 2017-11-24:
You can use the same principle for temperature as explained in the [wiki](https://github.com/SBFspot/SBFspot/wiki/Ambient-Temperature-PVoutput)
Like this (Frequency as V6)
`
CREATE VIEW vwPvoData AS
    SELECT dd.Timestamp, dd.Name, dd.Type, dd.Serial,  
        dd.TotalYield AS V1,
        dd.Power AS V2,  
        cons.EnergyUsed AS V3,
        cons.PowerUsed AS V4,  
        NULL AS V5,  
        spot.Frequency AS V6,  
        NULL AS V7,
        NULL AS V8,
        NULL AS V9,  
        NULL AS V10,
        NULL AS V11,
        NULL AS V12,  
        dd.PVoutput  
    FROM vwDayData AS dd  
    LEFT JOIN vwAvgSpotData AS spot ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min  
    LEFT JOIN vwAvgConsumption AS cons ON dd.Timestamp = cons.Nearest5min  
    ORDER BY dd.Timestamp DESC;
`

npkamen, 2017-11-24:
Thanks.  I'll give it a go.

npkamen, 2017-11-24:
I followed the guide (but slightly changing it in my case as I would still like temperature but don't need voltage).

First thing I did was drop the view.  

`DROP VIEW IF EXISTS vwPvoData;`

Then I tried to re-create it using this:

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              dd.Power AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Frequency AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;

But I get an error when issueing the create command.

`Error: table vwPvoData already exists`

Any ideas?

npkamen, 2017-11-24:
OK so I got it to work (by copying the sqlite database, working with it offline).

I tried to test the view afterwards and get the following error:

sqlite> select * from vwPvoData;
Error: no such column: spot.Frequency

Making progress I guess, just trying to fumble my way through working with databases etc.

Sorry for all of the silly questions.

npkamen, 2017-11-24:
OK so I got it to upload.  I had to alter the vwAvgSpotData view as well (it didn't have the frequency data in it).

Now it is uploading to PVOutput.  How do I change the column heading?  It is still titled voltage but the data is definitely frequency?

npkamen, 2017-11-25:
Just did some reading on PVOutput and can see that the v6 field has to be Voltage.  I've switched it back to Voltage and started sending V7 as Frequency (and donated and configured the custom field).

I'll see how long it takes for the frequency data to start appearing on PVoutput.

closed 2017-11-25 by npkamen