SBFspot/SBFspot #190: sbfspot 3.2.0 beta with Raspbian Stretch

cerberusde, 2018-01-16:
I use sbfspot 3.2.0 beta with the Multigat. I have updated my system to Raspbian Stretch and now I get the following message.

/home/pi/sbfspot/SBFspot: error while loading shared libraries: libboost_date_time.so.1.55.0: cannot open shared object file: No such file or directory

In Stretch the libboost-date-time-dev 1.62.0.1+b4 is installed.

 regards, cerberusde

sbf-, 2018-01-16:
You will need to recompile SBFspot and link with the new libraries

cerberusde, 2018-01-16:
Sorry, but I do not know how that works.

sbf-, 2018-01-18:
Well, you already did this before ;-)
`sudo make install_sqlite`
Just make sure you have installed all the dependencies (see documentation)

cerberusde, 2018-01-18:
Thanks, now it works.

closed 2018-01-18 by cerberusde