SBFspot/SBFspot #193: Migrating to new system

Alootje, 2018-03-01:
Because the system at my parents isn't running as supposed to due to an ISP change I'm planning to migrate their install to a new system. Current install is on a Raspberry Pi B and I want to install it on a banana pi which is currently their NAS.
Normally I think migrating is as easy as:
1. Exporting the MySQL database from RPi
2. Setup SBFspot on BPI
3. Import MySQL database from BPi and drink a beer
Correct?

___

Some background information: as of February 21st (the day of the ISP change) the data isn't written to PVoutput. The .sql files are still generated in the 'SBFspot' folder, but I don't know if all these values are written to the MySQL table (I didn't have time to check this). Because the .sql files are generated I asume the communication between convertor and Pi is okay...

When I run the upload daemon I receive the following errors:
```
Failed to get current plant time (-1)
SetplantTime returned an error: -1
```
Searching a bit shows that there could be a mismatch between the time of the Pi and the convertor, but both have the correct time already. Updating/switching between timezones and back didn't solve the issue.

sbf-, 2018-03-01:
That SetPlantTime isn't a big issue, especially when time is correct on inverter and host.
What is the output when you run the PVOutput view in MySQL?

Alootje, 2018-03-01:
How do I manage that? Is that a query via ssh? Or do I need to view a certain table in MySQL itself?

sbf-, 2018-03-02:
Execute this query in MySQL:
`SQL
select * from vwPvoData where PVoutput is null;`

Also, check your sbfspotupload.cfg or post it here (without API key)

Alootje, 2018-03-06:
Last week I had some time to work on this issue (system is located at my parents making troubleshooting more difficult). Turns out the database isn't managed by MySQL but by SQLite (leading me to think I've installed it according "[SBFspot V3 install on a Raspberry Pi EN V1.2](https://github.com/SBFspot/SBFspot/files/1696798/SBFspot.V3.install.on.a.Raspberry.Pi.EN.V1.2.pdf)". Apart from that tutorial my experience with SQLite is '0'.

Via WinSCP I scanned the directories on the Pi and found a 'logs' folder. This showed a lot 'couldn't resolve host name' errors. Although I updated the fixed IP in /etc/network/interfaces, I also needed to update /etc/resolv.conf. After I pointed this to the gateway IP SBFspot started working. So my parents are happy for now.

Still I would like to investigate how I can migrate to a new system. Do I need to:
1. Export (dump) the database from the RPi,
2. Setup SBFspot on Banana Pi  (since we only have one Bluetooth dongle. do I need to do this between sunset/sunrise so the current system keeps functioning?)
3. At the step "Create an empty database" Import the database on the BPI

Or can I just  copy the SBFspot.db from the RPi to the BPi and all is good?

sbf-, 2018-03-06:
It's as easy as copying SBFspot.db to your new system.

closed 2018-03-11 by sbf-