SBFspot/SBFspot #194: Compilation error

rucksman, 2018-03-03:
Have version 3.31 running on a pi3 without any problems sind more than a year. Wanted to migrate to another pi (pi2) and was pleased to see that there is a new version which has "Support for Smart Energy Inverters (Solar Inverters with battery)" (whatever that means because I have a Smart Energy Inverter since more than a year and was able to monitor it with version 3.31; maybe some short words on what has changed??).

My problem is that I cannot compile. I made the sbfspot.3 directory, fetched the latest SBFspot version und un-tared it. Then I changed into directory /home/pi(sbfspot.3/SBFSpot and made "make install_mysql". The error message I get (sorry for german language, but I am sure you get the point):
`test -d /usr/local/bin/sbfspot.3 || mkdir -p /usr/local/bin/sbfspot.3
cp mysql/bin/SBFspot /usr/local/bin/sbfspot.3
cp: Aufruf von stat für „mysql/bin/SBFspot“ nicht möglich: Datei oder Verzeichnis nicht gefunden
makefile:108: recipe for target 'install' failed
make: *** [install] Error 1
`

If I use the old version (3.31) just out of curiosity I do not get this message (but a different error but thats OK, I know why).

So what do I have to do to compile the MySQL Version of SBFspot 3.40?

sbf-, 2018-03-03:
Sorry for the missing documentation. It's a two step process now.
make mysql
sudo make install_mysql

rucksman, 2018-03-03:
Thanks, worked for me! Can you explain the changes regarding the Smart Energy Inverters? Thanks for your excellent work!

rucksman, 2018-03-03:
There is an error during "make install_mysql":
    cp SBFspot.cfg /usr/local/bin/sbfspot.3SBFspot.default.cfg

A slash is missing between the directory name and the cgf-filename.

sbf-, 2018-03-04:
Thanks, I'll fix it asap
The changes for SE inverters include the `vwBatteryData` where you should find some battery data like charging current and voltage, battery temperature ...

closed 2018-03-04 by sbf-

gibbedy, 2018-09-14:
guess why I'm here

commit e62de776 (2018-03-04): Fix #194 Error in makefile