SBFspot/SBFspot #195: SBFSpot refuses to Auto Start on reboot

fritserasmus, 2018-03-03:
So I installed "SBFspot_SRC_331_Linux_Win32.tar.gz" after my previous installation just stopped working :-(

I am unable to let this command run after reboot / power out
sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

I followed the steps in the manual and rechecked:
So this command works perfect if the pi was rebooted.
sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

I added the lines:

# Start SBFspotUploadDaemon
sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

in the /etc/rc.local  file, but when I reboot the Pi, no values are posted to PVOUTPUT

Only when I manually enter:   sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

the values start to appear in PVOUTPUT.

I have a brand new Raspberry Pi 3 - Model B with latest Rasbian installed.

Any pointers - please

sbf-, 2018-03-04:
Do you have an exit command before start of daemon?

fritserasmus, 2018-03-04:
Nope!

Thanks for that!.
I just edited the file.

I will reboot tomorrow morning to see if it started.

Thanks for your quick reply!!!.

Regards

Frits

fritserasmus, 2018-03-05:
thanks: It works 100% now

sbf-, 2018-03-06:
Nice to hear.

closed 2018-03-06 by sbf-