SBFspot/SBFspot #199: (Request) Is there a way to do a one off collection of all data stored in inverter?

SCUZNUTS, 2018-03-09:
I have a bluetooth inverter setup, humming along for the past year but I would like to collect the older data from more than a year ago.
Is there a one off 'Collect Everything'?

sbf-, 2018-03-09:
It depends on the inverter you have.
Daydata (every 5min) is kept for 62days
Monthdata (every day) is kept until ... (forever??) - Same for events
Spotdata is not recorded.
To get the older data, execute

> `SBFspot -ad62 -am15 -ae15 -sp0 -v`

SCUZNUTS, 2018-03-09:
Device Class:     Solar Inverters
Device Type:      SB 5000TL-21

Connecting over bluetooth. Tried the above command, it does go back to 01/01/2017 but not further.
I grabbed a laptop and installed 'Sunny Explorer' from SMA and the month data does populate from back into 2013/14.
I've used that to export the data.

Any other command to try for SBFspot?
Otherwise all good to mark this closed.

sbf-, 2018-03-09:
Well I thought it was obvious the 62 in -ad62 is for 62 days back, so the 15 in -am15 is for 15 months ;-)
The bigger the number, the mor you go back in time.

closed 2018-03-09 by sbf-