SBFspot/SBFspot #203: SBFspot Stopped SMA Portal Email Notifications?

bj-github, 2018-03-17:
Since firing up an instance of this a few days ago, I have stopped getting emails from the SMA Sunny Boy portal. When I finally had time to investigate within the portal a couple of days later, I noticed in the Portal's log the message "Virtual channels were edited by <myEmailAddress>" timestamped to my first run of SBFspot. That was followed by daily messages about the time being set in the interim, which I subsequently disabled in SBFspot since the WebConnect module in my SB 4000TL-21 is already doing that.

More curious is that my subsequent attempts to get emails from the portal, either scheduled *or* on demand, yields nothing in my inbox. So it appears to me that FSBspot has done something to cripple emails from the Sunny Boy portal. I'd appreciate any insights/help you can offer to restore my receipt of emails from the portal, please?

Apart from this, I have to say I was so happy to find it (hadn't looked for Linux-based software for this for several years, since first installing the panels), and am otherwise very happy with it. I had previously been relying on a 3rd party Android app that was prone to randomly stop working after anything from a few hours to a day or three. Especially annoying if I was away for a period!

With thanks for the work, and in anticipation of any help - Barry J.

sbf-, 2018-03-17:
It's the first time I hear about that and to be honest, I really don't have an idea...
Did you already try to stop SBFspot for 1 or 2 days to see if emails are being sent again?
I suppose you checked the portal config multiple times?

bj-github, 2018-03-19:
Thanks for your reply. Yes, I've been all through the SMA portal, disabling and enabling different reports, and requesting email copies as well as changing scheduled email times. Per your suggestion, I have disabled SBFspot for a few days. While it's disabled, I'll contact the portal folk to see if they can tell me what they see from their end. I'll certainly report back here with anything additional that I uncover.

bj-github, 2018-03-21:
Yesterday I fired up SBFspot again. And overnight the SMA portal emails started again. No messages in the SMA portal log in the interim: go figure!?

(Mainly restarted it after frustration in trying to connect to SMA portal folk to ask them what they saw on their end. But their request form excluded Australia as a location from which they take reports. And SMA Australia pointed to that form as a place to submit technical issues!? And a form on SMA Australia's website returned a "failed" error when I tried submitting a "general comment" to them seeking help about that. And their ph# doesn't take calls from individuals like me. <sigh>)

closed 2018-03-21 by bj-github

sbf-, 2018-03-21:
Glad this is "fixed" ;-)