SBFspot/SBFspot #204: Unknown Inverter Type warning in SBFspot V3.3.1

fshumeli, 2018-03-17:
Hello.
I've just discovered SBFspot and I've installed V3.3.1 on Raspberry Pi 3, running Raspbian.
Bluetooth disabled (my SMA inverted doesn't have BT) and ethernet configured.
All's working fine and testing `/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv` works well... except I get a little warning:

> Logon OK
> Unknown Inverter Type. 
> Report this issue at https://sbfspot.codeplex.com/workitem/list/basic with following info: 0x0000246A and Inverter Type=<Fill in the exact type> (e.g. SB1300TL-10)

So here I am, reporting the error, if it's of any use.
My inverter type is SMA SunnyBoy V5.0 aka SB5.0-1AV-40

I don't think the warning has any impact on the script as it ends with :
> Sat Mar 17 13:39:13 2018: INFO: Done.

All values are exported to database as well.

Anyway, thanks for any feedback on this.

sbf-, 2018-03-17:
This has no impact on correct funtioning, but to fix it you can download the TagList.txt files from the sources: https://github.com/SBFspot/SBFspot/blob/master/SBFspot/TagListEN-US.txt

fshumeli, 2018-03-18:
Actually, even after replacing the old TagListEN-US.txt & TagListFR-FR.txt with the new ones, it didn't change a thing. 
My settings in `/usr/local/bin/sbfspot.3/SBFspot.cfg` are set to` Locale=fr-FR`
I switched to `Locale=en-US` but still no luck.
I know it doesn't really matter, but I'm wondering how to make it "clean"...

thanks

sbf-, 2018-03-18:
Did you replace them in `/usr/local/bin/sbfspot.3` ?
It should work, because `SB5.0-1AV-40` is defined in all taglists: `9322=SB5.0-1AV-40\0\Sunny Boy 5.0`

fshumeli, 2018-03-18:
I've located it in the taglist indeed, so I wasn't sure why it didn't work... Until you made me realize I overwrote those in `/home/pi/sbfspot.3/SBFspot` !!! 
I'm on my way to move them to `/usr/local/bin/sbfspot.3` ! 
Sorry for the trouble & thanks a lot !

fshumeli, 2018-03-18:
Works like a charm now.
Thanks again !

sbf-, 2018-03-18:
You're welcome!

closed 2018-03-18 by sbf-