SBFspot/SBFspot #205: How do you deal with multiple inverters? 20000TL shows up as 2x BT devices 10KW each

michaelvorst, 2018-03-19:
Just got SBFspot 3.3.1 working on RPI, but have discovered that the SMA 20000TL presents itself as two separate BT devices with different SNs of 10KW each.

_Can you please let me know how to get more than one set of SBFspot data out to PVOutput?_ The SBFspot manual only really covers a single inverter setup.

I've tried adding the second BT address in the SBFSpot.cfg, but after that, the CSV only contains the data from the first BT device listed. I also changed the MIS_enabled to 1, but this didn't seem to do anything.

I had added the second inverter ID in the SBFspotUpload.cfg, but I suspect that this is only going to do something if the first stage of data collection is working correctly.

( I'm using 3.3.1 as 3.4.0 as still returns makefile errors during install, and I don't know enough about Linux to figure out how to correct it).

sbf-, 2018-03-19:
This is an interesting case. If you want, you can send me (s.b.f at skynet dot be) a debug log (-v5 -d5)

If `MIS_enabled` won't work, you can create 2 config files. Each config has its proper BT address and plantname.
Call SBFspot twice with each config:
`SBFspot -v -cfg2110564778.cfg`
`SBFspot -v -cfg2110563802.cfg`

For the daemon config, enter the 2 serial numbers with PVoutput System ID's
`PVoutput_SID=2110564778:<sid1>,2110563802:<sid2>

For V3.4 there is a 2-step build process. See [wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite) for more info.

michaelvorst, 2018-03-20:
I created the two named files, gave each one the matching BT address for the inverter SN, and placed them in /usr/local/bin/SBFspot.3. However, when I tried to start the process it returned the following error:

[image]

The error suggested the process was lookingo for a filename without "CFG..." at the start, so I edited the file to remove that, and the error message changed slightly to the second error in the screenshot.

Any ideas? Thanks for helping, BTW...

sbf-, 2018-03-20:
😃  I see, the file is named `cfg2110563802.cfg` so the command line should be
`SBFspot -v -cfgcfg2110563802.cfg`

`-cfg`is the argument name

BTW, 2 Rpi's is overkill

michaelvorst, 2018-03-21:
Hi,

Got it working in the end - thanks for your help.

https://pvoutput.org/intraday.jsp?id=55687&sid=57888 - this is a parent of
the 57886 and 57847 individual inputs.

Initially, I got the same error with the renamed files:

They are accessible - I can sudo nano them and they are opening fine. I
ended up copying them so there are two sets, and that seems to work:

The rest of the setup with the daemon worked without issue.

closed 2018-03-24 by sbf-