SBFspot/SBFspot #207: How to collect battery statistics with v3.4.0

GundarooSolar, 2018-03-24:
Hi SBF,

Just upgraded to version 3.4.0 as I was excited to see that you have added battery statistics into the release.

However, I cannot see any battery data being collected for my Sunny Island 8.0, and in fact the output of the SBFspot exe no longer outputs the battery charge level (which it used to do).

I used to see this information in the output from SBFspot
_SUSyID: 289 - SN: 1261038518
Device Name:      SN: 1261038518
Device Class:     DevClss7
Device Type:      Sunny Island 8.0H
Software Version: 01.03.01.R
Serial number:    1261038518
SUSyID: 289 - SN: 1261038518
Batt. Charging Status: 96%
SUSyID: 289 - SN: 1261038518
Batt. Temperature: 26.0°C
Batt. Voltage    : 51.60V
Batt. Current    : 0.000A_

After the upgrade (compiled and installed the binaries, and updated the DB to have the additional table and view), I now see only the following information:

_SUSyID: 289 - SN: 1261038518
Device Name:      SN: 1261038518
Device Class:     DevClss7
Device Type:      Sunny Island 8.0H
Software Version: 01.03.01.R
Serial number:    1261038518_

Is there something I need to do to get the program to query and store the battery data?

Thanks.

sbf-, 2018-03-27:
Thanks for reporting this.
I tried to fix it in the `Master` branch. Please download & test it.

GundarooSolar, 2018-03-30:
Thanks SBF. Have loaded the new code and it is working just fine (including storing the data into the database. So, bug fixed! Cheers!

Now I just need to work out how to get it sent through to PVoutput

Cheers,
Danny.

sbf-, 2018-03-30:
> Now I just need to work out how to get it sent through to PVoutput

Have a look at the "Tweaking" section in [this wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)

sbf-, 2018-07-19:
Fixed in 3.4.1

commit d9bd145f (2018-03-27): Fix #207: How to collect battery statistics with v3.4.0
closed 2018-07-19 by sbf-