SBFspot/SBFspot #208: interfacing with Home Automation instead of PVoutput ?

fshumeli, 2018-03-24:
Hello
Not really an issue, but more of a feature request/suggestion...

I'm having trouble figuring out what would be the best way to extract some data from my inverter to make it available for my home automation system. 
I actually use [HomeBridge](https://github.com/nfarina/homebridge) on a raspberry Pi along with my Fibaro Home Center Light box. 
Homebridge could probably read data and send it to as a "virtual device" to my Home app. (this I also need to figure out yet. writing a simple plugin for HomeBridge is beyond my skills, but I'm working on it)

Let's say I'd like to display Today's Energy Production (_EToday_). 
Should I have an SQL request in a crontab that extracts EToday every few minutes and writes it in a text file (overwriting itself each time), then find a way to tell Homebridge to read from this file ?
Any better idea ?

Thanks for any advice or help with this.

Regards from Brussels, BE.

sbf-, 2018-03-27:
You can create a simple bash script with `tail -n 1 spotfile.csv` to get the last line and filter out EToday
This script can be launched every time when SBFspot has run (no extra cron job)

rstrrt, 2018-03-30:
Hi,

could you please elaborate on how to make SBFSpot call a script each time new data is added to the csv?
I think this approach could work well for publishing generation data to my mqtt broker (emoncms running on emonpi).
Thanks a lot!

rob

sbf-, 2018-03-30:
See "Automation" in [this wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)
Call your script at the end of the `daydata` script

closed 2018-06-16 by sbf-

sbf-, 2019-08-17:
For mqtt, see also #248