SBFspot/SBFspot #209: Consumption data upload

bshaffer82, 2018-03-27:
I've created a consumption meter which can provide usage watts using a spark particle.  I've had SBFspot working for a couple of years now and I want to add the consumption data so it gets uploaded to PVOutupt.  

What do I need to do?
Can I just add the data into the consumption mysql table?  if so what's the required table format?  Do I need to tag the data with the same timestamp as the data coming out of Bluetooth data capture?

I'm an intermediate programmer but not that experienced with sqlite.

Thanks

sbf-, 2018-03-27:
Yes, fill consumption table with Timestamp, Energy (Wh) and Power (W)
The only challenge is the upload when it's dark and there is no production. Upload stops when production stops, so this is still an issue. This requires changes to the code.

bshaffer82, 2018-03-31:
What is the format for those three?  

Specifically, what is the date format?  I would rather push the data from python than try an recompile SBFspot.

sbf-, 2018-04-02:
date is [UTC](https://www.epochconverter.com/) stored as int(4)
power in W
energy in Wh

Bruut8, 2019-09-28:
Will this work when adding -finq to the daydata cron job? That should force the uploads to go on during the night right?

Bruut8, 2019-10-03:
To answer my own comment for those who need the info:

-finq does not force uploads during the night. 

The script stops updating data in the 'daydata' table when there is no more energy generated. Without data in that table, the 'vwpvodata' is not populated, hence the uploadscript has no data to upload.
To upload consumption data 24/7 you need to create a script to calculate the power consumption data and add that in the 'consumption' table every five minutes. As mentioned above, timestamp is in unixtime, power in W and energy in Wh. The energy value is cumulative (just like 'totalyield' in energy generation), pvoutput will automatically calculate the power generated between datapoints.
This data will then appear in the 'vwconsumption' view and (during daylight) in the 'vwpvodata'  view, where the 'nearest5min' is added. Alas, this is not happening when it is dark, because there is no data in 'daydata'.
The last step then is adding data in the 'daydata'  table manually when there is no output from the inverter. Make sure the timestamp is the same as the 'nearest5min' mentioned before, set the correct serialnumber and put 0 in energy- and powergeneration. Leave the pvoutput column at null (0 will not work).
The above step will ensure the 'vwpvodata' table is filled and uploaded. 

If there is an easier way please let me know, until then the above step will solve the problem without modifying the SBFspot code.

aradovan, 2020-02-24:
> Yes, fill consumption table with Timestamp, Energy (Wh) and Power (W)
> The only challenge is the upload when it's dark and there is no production. Upload stops when production stops, so this is still an issue. This requires changes to the code.

Hi, firstly sorry about my english... I would like to get the current consumption as I do in sunny portal, but running "SBFspot -v -finq -nocsv" I'm unable find that value into the text result. I've also search for it in vwAvgConsumption table (also vwPvoData that joins it) but this table is empty. You tell about filling this table but my problem is how to get that consumption value to fill it or get it in text result... Is there some [-option] for showing it? I dont need to upload it, just read it each time I run the command...