SBFspot/SBFspot #212: Unable to send battery statistics to PVOutput for my SunnyIsland

GundarooSolar, 2018-04-07:
Now that you have added the great functionality of pulling battery metrics (THANK YOU!!!), I'm trying to get the data sent to PVOutput to be displayed along side the power generation data.

My solar config is as follows:
   2 x SunnyBoy 5000 units connected to 3 strings of panels
   1 x SunnyIsland 8.0 as the battery inverter, which also controls power to the house
   Off grid, so no grid connection data.

I'm happily collecting and presenting the power generation data from the SunnyBoy units on PVOutput (https://pvoutput.org/intraday.jsp?id=53020&sid=53006)

What I'm now trying to do is add in some custom metrics for the battery state onto the graph.

The main hurdle I'm hitting is that, because the SI 8.0 is not a power generation inverter, it is not generating any data in the DayData table (when I trace what is going on, the collection of day day gives me nothing from this unit). 

Given that the DayData table is the main driver behind what is being sent to PVOutput, it seems that I'll never get any data sent for this unit. 

One thought was to expand the vwPvoData view to join to the vwBatteryData view based on the timestamp only (ignore the serial number) and effectively send the Battery info as part of the info for both of the SunnyBoy systems. 

Is this a good way to go about it, or am I just being dense and have missed an obvious configuration that would allow me to get my Batt data up to PVoutput?

Note that on the PVOutput site I have added the SI as a new device and for some reason it managed to send some data for 2 days in the past, but believe that is an aberation and not how it is functioning now. 

Cheers,
Danny.

GundarooSolar, 2018-04-10:
Yeah - that is what I suspected.
The device doesn't seem to have as much capability to output things like 5 min data etc as compared to the SunnyBoys (which seem more modern). 

Some of the newer battery inverters may, but the Sunny Island doesn't appear to. Even when I use the SMA supplied Sunny Explorer it only gets daily/monthly files from the SB units.

That is why I was thinking of changing the vwPvoData to link in the battery metrics to appear to belong to the Sunny Boy units. That seemed the smallest hack to how it currently works to get the data sent. 

I've attached the output from the operation of SBFspot in full logging/debug mode.

The device with IP address 192.168.1.190 is the Sunny Island. The other two (191, 192) are the SB units.

[output.log](https://github.com/SBFspot/SBFspot/files/1894429/output.log)

sbf-, 2018-07-22:
I'm afraid this is not feasable the way SBFspot is designed.
If you have a workaround, please let us know.

closed 2018-07-22 by sbf-