SBFspot/SBFspot #214: MonthData DB duplicates

MatteoVentura, 2018-04-19:
Hey awesome work! 

Im just having some weird duplicates in my database. here's an example.
So I have 2 different transformers from who I fetch data.

[image]

As you see it's the same, just 5-minute difference.

Crontab cfg:

[image]

Monthdata.sh cfg:

[image]

Config: https://pastebin.com/mU536JTf

sbf-, 2018-04-20:
That’s strange indeed. That was a bug in the past, and it was solved by deleting existing records for that month. In theory they should disappear while importing that month again.

MatteoVentura, 2018-04-22:
Any tips ? What I can try ?

sbf-, 2018-04-23:
Since monthdata is imported once every day, the duplicates should disappear automatically.

MatteoVentura, 2018-04-23:
Yes,  I understand but why does it sometimes  take 21:55 and other times 22:00?

Verzonden van mijn Redmi Note 4

sbf-, 2018-04-23:
I really don't know, but this date/time is reported by your inverter. Maybe SBFspot should round these to the nearest hour.

closed 2018-07-19 by sbf-