SBFspot/SBFspot #215: start SBFspotUpload.service failed because the control process exited with error code.

peterjgithub, 2018-04-28:
I don't really know where to search for fixing.
systemctl status SBFspotUpload.service returns:

SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2018-04-28 12:03:23 CEST; 55s ago
  Process: 10340 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon -p /run/sbfspotupload.pid 2>&1> /dev/null (code=exited, status=1/FAILURE)

Any suggestion where I can start?

using: Raspberry Pi 3 Model B+ installed with openHABianPi

peterjgithub, 2018-04-28:
I could resolve the issue.

'cd /etc/sbfspot/smadata
mkdir logs`  

Now it is working & generating logfiles.

I didn't close this issue, so that devs can evaluate if a fix is needed. Feel free to close it when this is considered.

sbf-, 2018-04-28:
I'll leave it open for now.
Thinking about creating the log folder when it doesn't exist.
Thanks for reporting.

sbf-, 2018-06-14:
IT seems like I left some buggy testcode in upload daemon of v3.4.1
I removed this release and come back later on this .

sbf-, 2018-07-22:
Fixed in V3.5.0

closed 2018-07-22 by sbf-