SBFspot/SBFspot #216: Unable to open output file. See syslog for details.

shadi1989, 2018-04-29:
I've re-installed SBFspot on my Raspberry Pi to upload my data to PVOutput but it seems that there's an error somewhere.

I used the manual "SBFspot.V3.install.on.a.Raspberry.Pi.EN.V1.2.pdf"
I'm at:
Testing
Start the daemon:
/usr/local/bin/sbfspot.3/SBFspotUploadDaemon
If all goes well, you see nothing on the commandline but there should be a logfile in
/home/pi/smadata/logs:
[09:49:47] INFO: : Starting...
[09:49:47] INFO: : Uploading 30 datapoints, starting with

When i run the "/usr/local/bin/sbfspot.3/SBFspotUploadDaemon" i see no error in commandline but when i open the log file i see:
[23:58:47] INFO: Starting SBFspotUploadDeamon Version 1.0.1
[23:58:47] ERROR: Unable to open output file. See syslog for details.

Anyone got an idea how i can fix this?

sbf-, 2018-04-30:
What’s in syslog?
It looks like an error in this doc. You should ‘sudo’ to start the daemon.
`sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon`

shadi1989, 2018-04-30:
@sbf- 
I've added in the SBFspotUploadDaemon in rc.local
"#Start SBFspotUploadDaemon
sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon"

Checked the logs today and it was filled with uploads so guess it works. i did not edit anything after i make the Issue on Github so it's kinda strange.