SBFspot/SBFspot #217: 1 inverter data not uploading

peterjgithub, 2018-04-30:
Hi.

2 inverters are uploading well:
https://pvoutput.org/list.jsp?id=65964&sid=58683
https://pvoutput.org/list.jsp?id=65964&sid=58680

1 inverter is not uploading for some reason:
https://pvoutput.org/list.jsp?id=65964&sid=58681
(the data you see is uploaded using another tool: Oxley)

Data is received from the inverter: I attached a logfile ( sudo /usr/local/bin/sbfspot.3/SBFspot -v5 -ad5 -am5 -ae5 -finq.txt)
[sudo :usr:local:bin:sbfspot.3:SBFspot -v5 -ad5 -am5 -ae5 -finq.txt]

Also find my cfg file attached
(adapted: API key masked and .cfg > .txt)
[SBFspotUpload.txt](https://github.com/SBFspot/SBFspot/files/1960383/SBFspotUpload.txt)

I tried with LogLevel=debug, but that didn't learn me a lot.

Any advice?

sbf-, 2018-04-30:
This is a limitation of PVoutput: [(source)](https://pvoutput.org/help.html#api-addbatchstatus)
> **Restrictions and Limitations**
> The date parameter must be not be older than 14 days from the current date.
> **Donation Bonus Features**
> The date parameter must be not be older than 90 days from the current date.

You can upload the missing days via [PVoutput's CSV uploader](https://pvoutput.org/load.jsp)

peterjgithub, 2018-05-03:
It worked fine. All data have been processed at pvoutput.org.
I created a team to aggregate all historic data. It took a couple of days, but these calculations ended up all perfectly: https://pvoutput.org/listteam.jsp?tid=1349&t=y .
Thanks for supporting.

closed 2018-05-03 by peterjgithub

pagaille, 2022-06-23:
Strange. I used the latest version and used various combination of -ad and -startdate. The data is retrieved but not uploaded to PVoutput except the current day. How does the upload daemon know what records needs to be uploaded ? NB : I'm using a remote MariaDB server.

pagaille, 2022-06-23:
Thanks. The DB is indeed missing the dates I want to be uploaded. We jump from today 2022-06-23 straight to 2022-06-19.
When running` SBFspot -v5 -nocsv -ad4 -am0 -ae0 -sp0  -cfg/home/pi/SBFspot.cfg`, the log says ;
```
<snip>
        Efficiency  :    0.00%
SUSyID: 99 - SN: 2001680707
Grid Freq. : 50.01Hz
SUSyID: 99 - SN: 2001680707
Current Inverter Time: 23/06/2022 19:59:20
Inverter Wake-Up Time: 23/06/2022 19:59:20
Inverter Sleep Time  : 23/06/2022 19:59:20
********************
* ArchiveDayData() *
********************
startTime: 23/06/2022 00:00:00
Overwriting existing record: 23/06/2022 15:05:00 - 908.315kWh - 1728W
SUSyID: 99 - SN: 2001680707
23/06/2022 05:50:00 : 900.732kWh - 0.000W
23/06/2022 05:55:00 : 900.732kWh - 0.000W
23/06/2022 06:00:00 : 900.732kWh - 0.000W
<snip>
23/06/2022 19:20:00 : 910.634kWh - 132.000W
23/06/2022 19:25:00 : 910.646kWh - 144.000W
23/06/2022 19:30:00 : 910.659kWh - 156.000W
23/06/2022 19:35:00 : 910.668kWh - 108.000W
23/06/2022 19:40:00 : 910.675kWh - 84.000W
23/06/2022 19:45:00 : 910.681kWh - 72.000W
23/06/2022 19:50:00 : 910.687kWh - 72.000W
23/06/2022 19:55:00 : 910.693kWh - 72.000W
======
********************
* ArchiveDayData() *
********************
startTime: 22/06/2022 00:00:00
********************
* ArchiveDayData() *
********************
startTime: 21/06/2022 00:00:00
********************
* ArchiveDayData() *
********************
startTime: 20/06/2022 00:00:00
Thu Jun 23 19:59:23 2022: INFO: Done.

```