SBFspot/SBFspot #224: SBFSpotUpload.cfg - log level doesn't work

mkratz, 2018-06-14:
I'm attempting to reduce logging on my raspberry pi, in order to reduce disk writes.

I found in SBFspotUpload.cfg there is a LogLevel option, however it doesn't seem to work:

```
#LogLevel=debug|info|error (default info)
LogLevel=error
```

I'm still getting INFO logs:

```
[16:30:21] INFO: Uploading datapoint: 20180614,16:30,47318084,48,,,,243.36,224.39,298.26,20,13,48.9 => OK (200)
[16:34:39] INFO: Stopping Daemon...
[16:34:43] INFO: Starting SBFspotUploadDaemon Version 1.1.0
[16:34:43] INFO: Starting Daemon...
[16:35:31] INFO: Uploading datapoint: 20180614,16:35,47318087,36,,,,243.38,224.39,283.8,25,16,48.8 => OK (200)
[16:40:31] INFO: Uploading datapoint: 20180614,16:40,47318088,12,,,,243.97,207.43,283.78,12,7,48.8 => OK (200)
```

sbf-, 2018-06-14:
That's correct, it's not yet implemented.

closed 2018-07-22 by sbf-

sbf-, 2018-07-22:
Fixed in V3.5.0

commit e9c50a05 (2018-07-22): Fix #224: SBFSpotUpload.cfg - log level doesn't work