SBFspot/SBFspot #225: Backfill to pvoutput

mk-au, 2018-06-14:
I have a question about backfill capabilities to pvoutput.

If SBFSpot is offline for days or weeks, will it grab all missing data from the inverter (assuming the inverter stores historical data - e.g recent SMA bluetooth inverters) and backfill to pvoutput?

sbf-, 2018-06-14:
It will, but semi-automatic...
You will have to run SBFspot manually with -dxx (xx=1 to ... days - limit is 300 for testing purposes)
Daemon will upload the data automatically.

There are some limits applied:
- Inverter stores 62 days of daydata(every 5 minutes)
- PVoutput allows 14days of backload (unless you're a donator, then it's 90 days)

mk-au, 2018-06-14:
Thanks for the info @sbf-  That's useful to know. The previous program I was using would auto-backfill. Wasn't sure if the capability was there in SBFSpot. Glad to know its there at least in a manual capacity if needed.

Is the Daemon aware of donation status? i.e. so it knows it can backload up to 90 days?

sbf-, 2018-06-14:
"Auto-backfill" is on the todo list...
Yes, daemon is donation status aware

mk-au, 2018-06-15:
Thanks, good to know.

closed 2018-06-15 by mk-au

rdevries1965, 2019-04-19:
> It will, but semi-automatic...
> You will have to run SBFspot manually with -dxx (xx=1 to ... days - limit is 300 for testing purposes)
> Daemon will upload the data automatically.
>
> There are some [...]

Simple question: how do i run SBFspot manually in my putty shell using the -d90 command 
?

sbf-, 2019-04-20:
> how do i run SBFspot manually in my putty shell using the -d90 command

Your inverter stores only 62 days of data

```
/usr/local/bin/sbfspot.3/SBFspot -ad90 -finq -v
```
For more parameters, use SBFspot -?