SBFspot/SBFspot #226: SBFspotUploadDaemon does not need to run as root

freekdk, 2018-06-16:
Using the systemd service to start SBFspotUploadDaemon, this program now runs as root. This is not necessary. One of the problems I see is that the two programs require the config file (*.cfg) to be in the same folder as the program. Following common practices in linux is that configuration files are in /etc, or in some folder of /etc, or, if not there, in a local folder of the account the program runs in, like ~/.config/sbfspot or even ~/.sbfspot, where the program searches these locations for the configuration file. This also makes it possible to use User=pi in the SBFspotUpload.service file. Another problem in that case is that the account pi does have write access to /run. However this can be solved by not using /run/sbfspotupload.pid to store the pid, but something like /home/pi/sbfspotupload.pid.

sbf-, 2018-06-17:
> Using the systemd service to start SBFspotUploadDaemon, this program now runs as root. This is not necessary

Yes, I know since a few weeks ([See this thread)](https://www.zonstraal.be/forum/viewtopic.php?f=38&t=21181&p=247923#p247923)
I only need to update the wiki.

> One of the problems I see is that the two programs require the config file (*.cfg) to be in the same folder as the program.

Configuration file and executable in same directory is the default but SBFspot has a -cfg command line option to specify the path to the configuration file.
Same for the upload daemon: adding -c /etc/SBFspotUpload.cfg will do its job (Ok, it's undocumented - I'll add it to the wiki)

> Another problem in that case is that the account pi does have write access to /run.

The best place to write the PID file is /run/user/$UID - Problem for now is the upload daemon can't expand environment variables, so you'll have to hardcode it to /run/user/1000 

A better .service file would be:
```
[Unit]
Description=SBFspot Upload Daemon
After=network.target
[Service]
User=pi
Type=forking
TimeoutStopSec=60
ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon -c /etc/SBFspotUpload.cfg -p /run/user/1000/sbfspotupload.pid 2>&1> /dev/null
PIDFile=/run/user/1000/sbfspotupload.pid
Restart=no
RestartSec=300
SuccessExitStatus=SIGKILL
[Install]
WantedBy=default.target
```

sbf-, 2018-06-17:
wiki updated

closed 2018-06-17 by sbf-