SBFspot/SBFspot #227: Timezone support for SBFspotUpload

seriouswebby, 2018-06-28:
I have a MySQL database is running in the default UTC timezone, SBFspot is configured to my local timezone, when uploading my data to PVOutput the timezone remains as UTC. Does the upload service support timezone conversion?

Ive currently hacked the `vwDayData` view:
```shell
CREATE OR REPLACE VIEW vwDayData AS
	select CONVERT_TZ(From_UnixTime(Dat.TimeStamp),'SYSTEM','Europe/Brussels') AS TimeStamp,
	Inv.Name, Inv.Type, Dat.Serial,
	TotalYield,
	Power,
	PVOutput FROM DayData Dat
INNER JOIN Inverters Inv ON Dat.Serial=Inv.Serial
ORDER BY Dat.Timestamp Desc;
```

Thanks!

sbf-, 2018-06-29:
> Does the upload service support timezone conversion?

No, there was no reason for until now. Why is MySQL using UTC instead of local time?

seriouswebby, 2018-07-02:
Just using a docker image for [mysql-server](https://github.com/mysql/mysql-docker) and there wasnt any timezone settings out of the box, also I havent needed to alter the timezone.

Also my hack above caused the daemon to upload the same data each time it ran :(
```
[21:33:31] INFO: Uploading 30 datapoints, starting with 20180629,07:55,59395688,0,,,,0.00 => OK (200)
[21:34:30] INFO: Uploading 30 datapoints, starting with 20180629,07:55,59395688,0,,,,0.00 => OK (200)
[21:34:32] INFO: Uploading 30 datapoints, starting with 20180629,07:55,59395688,0,,,,0.00 => OK (200)
...
```