SBFspot/SBFspot #229: enhanement about funtion "archive events"

silliewimons, 2018-07-06:
at present one can only read one type of events at  time
- standard usage for "-ae" is to read the "user events"
- the parameter -installer allows to select "installer events"  but then user events can not be extracted

demand for enhancement
- provide an additional parameter example "-au" default "-au0"
  that parameter indicates that together wih the installar events also user events are to be read
  advantage - no need to execute two commands to read all events

kr willie

sbf-, 2018-07-09:
When you take a look at the code you’ll see user events are always read, regardless of the user/installeren login:

``` c++
for (int m = 0; m < cfg.archEventMonths; m++)
	{
		if (VERBOSE_LOW) cout << "Reading events: " << to_simple_string(dt_utc) << endl;
		//Get user level events
		rc = ArchiveEventData(Inverters, dt_utc, UG_USER);
		if (rc == E_EOF) break; // No more data (first event reached)
		else if (rc != E_OK) std::cerr << "ArchiveEventData(user) returned an error: " << rc << endl;

		//When logged in as installer, get installer level events
		if (cfg.userGroup == UG_INSTALLER)
		{
			rc = ArchiveEventData(Inverters, dt_utc, UG_INSTALLER);
			if (rc == E_EOF) break; // No more data (first event reached)
			else if (rc != E_OK) std::cerr << "ArchiveEventData(installer) returned an error: " << rc << endl;
		}

		//Move to previous month
		if (dt_utc.month() == 1)
			dt_utc = gregorian::date(dt_utc.year() - 1, 12, 1);
		else
			dt_utc = gregorian::date(dt_utc.year(), dt_utc.month() - 1, 1);

	}
```

There is probably een issue when you reach the very first user event and break out of the loop without querying the rest of the installer events.

sbf-, 2018-07-21:
@silliewimons can you elaborate please?
When running as installer, all events of user/installer should be retrieved.

sbf-, 2018-07-22:
"user" can't read "installer" events, the opposite is possible

closed 2018-07-22 by sbf-