SBFspot/SBFspot #231: CRITICAL: Logon failed. Check 'USER' Password

poune, 2018-07-23:
Logon failed on SB4.0 with speedwire.  The password is correct cause it work 1 on 20 !!!????

This is my logs : 

SBFspot V3.5.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -d5 -ad1 -am0 -ae0
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=192.168.1.249
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.8
Longitude=4.33
Timezone=Europe/Brussels
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Mon Jul 23 17:27:28 2018: INFO: Starting...
sunrise: 03:57
sunset : 19:40
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 928804231 (0x375C6D87)
Inverter IP address: 192.168.1.249 from SBFspot.cfg
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 26 00 10 60 65 09 A0 
00000020: FF FF FF FF FF FF 00 00 7D 00 
00000030: 87 6D 5C 37 00 00 00 00 00 00 
00000040: 01 80 00 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 00 00 00 
58 Bytes sent to IP [192.168.1.249]
ethGetPacket()
MAX_CommBuf is now 98 bytes
Received 98 bytes from IP [192.168.1.249]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 4E 00 10 60 65 13 A0 
00000020: 7D 00 87 6D 5C 37 00 00 56 01 
00000030: DD EE BB 76 00 00 00 00 00 00 
00000040: 01 80 01 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 03 00 00 00 FF 
00000060: 00 00 C0 FD 01 60 01 00 56 01 
00000070: DD EE BB 76 00 00 0A 00 0C 00 
00000080: 00 00 00 00 00 00 03 00 00 00 
00000090: 01 01 00 00 00 00 00 00 
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 13 A0 7D 00 87 
00000010: 6D 5C 37 00 00 56 01 DD EE BB 
00000020: 76 00 00 00 00 00 00 01 80 01 
00000030: 02 00 00 00 00 00 00 00 00 00 
00000040: 00 00 03 00 00 00 FF 00 00 C0 
00000050: FD 01 60 01 00 56 01 DD EE BB 
00000060: 76 00 00 0A 00 0C 00 00 00 00 
00000070: 00 00 00 03 00 00 00 01 01 00 
00000080: 00 00 00 00 
<<<=================================>>>
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 22 00 10 60 65 08 A0 
00000020: FF FF FF FF FF FF 00 03 7D 00 
00000030: 87 6D 5C 37 00 03 00 00 00 00 
00000040: 02 80 0E 01 FD FF FF FF FF FF 
00000050: 00 00 00 00 
54 Bytes sent to IP [192.168.1.249]
logonSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 3A 00 10 60 65 0E A0 
00000020: FF FF FF FF FF FF 00 01 7D 00 
00000030: 87 6D 5C 37 00 01 00 00 00 00 
00000040: 03 80 0C 04 FD FF 07 00 00 00 
00000050: 84 03 00 00 00 10 56 5B 00 00 
00000060: 00 00 B8 B8 B8 B8 B8 88 88 88 
00000070: 88 88 88 88 00 00 00 00 
78 Bytes sent to IP [192.168.1.249]
ethGetPacket()
Timeout reading socket
No data!
Mon Jul 23 17:27:33 2018: CRITICAL: Logon failed. Check 'USER' Password

What's wrong?

Thanx

sbf-, 2018-07-23:
No idea... password is OK, it's a communication issue.
Do you  have other SMA devices in your network?
Try with IP_Address=0.0.0.0 in cfg file

sbf-, 2018-07-26:
Does it work with sunny explorer (on windows)?

poune, 2018-07-27:
I don't have Windows, only Linux ... The web interface is OK ... I tested with the option -installer and -password:xxxx and same result, it pass sometimes with this infos : 

Device Name:      SB4.0-1AV-40 869
Device Class:     Onduleur solaire
Device Type:      Sunny Boy 4.0
Software Version: 01.03.17.R

but in processing no temperature : Device Temperature: 0.0°C

SpoBo, 2018-08-05:
Hi @poune I had a similar issue. Are you sure the mac of the BT device is correct ? hcitool scan showed 2 addresses for my inverter. One constantly gave the "Check 'USER' password" error. The other one worked perfectly fine.

poune, 2018-08-06:
Hi, I'm connecting with Speedwire only : no BT on this Sunny Boy ...

andpierard, 2019-04-01:
Hi, I got the same problem today trying to install SBFspot on a Raspberry 3pi with Speedwire option (SB2.5 inverter has no bluetooth).  Did you find an issue on your installation ?
Thanks a lot

poune, 2019-04-02:
Hello, no results with SBFspot so I activate modbus on the Sunny Boy and I get values with mbpoll (https://github.com/epsilonrt/mbpoll) with TCP.

andpierard, 2019-04-02:
Hello, Thank you for your reply. I'm afraid that SBFspot is not working so I should have to investigate the alternative solution you propose. Regards.

sbf-, 2019-04-02:
SBFspot works, but it's not bullet-proof as they say...
A known issue is when you have an SMA Energy Meter in your network. Don't know if this is the case?

andpierard, 2019-04-02:
Hello, Thank you for your reply. 
I was afraid that SBFspot will not work definitevely and so I will have to investigate the alternative solution you propose. 
And ...at last I put the User password (and not the Installer or administrator password) in the SBFspot.cfg file ... and it works !!!!!!!!!!!!! 
Regards.

andpierard, 2019-04-03:
> 
> 
> SBFspot works, but it's not bullet-proof as they say...
> A known issue is when you have an SMA Energy Meter in your network. Don't know if this is the case?

Hi SBF,
No, there is no SMA Energy Meter installed.
Thanks for the info.
Rgds

lesleypersyn, 2019-05-26:
If there are special characters in the password then you get the error. If you only user letters and numbers it's working fine.

closed 2019-07-28 by sbf-