SBFspot/SBFspot #232: SBFspotUploadDaemon[1165]: Unable to write to logfile

Downhilll, 2018-07-29:
I have a issue for de SBFspot upload Daemon, i read this on may raspberrypi
what do i wrong?

 pi@raspberrypi:~ $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-07-29 15:18:23 UTC; 2min 46s ago
 Main PID: 1165 (code=exited, status=1/FAILURE)

jul 29 15:18:23 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
jul 29 15:18:23 raspberrypi SBFspotUploadDaemon[1165]: Unable to write to logfile [/home/pi/smadata/logs/SBFspotUpload20180729.log]
jul 29 15:18:23 raspberrypi systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
jul 29 15:18:23 raspberrypi systemd[1]: SBFspotUpload.service: Unit entered failed state.
jul 29 15:18:23 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.

arnocs, 2018-07-29:
The SBFspotUpload runs as a user, that user does not have write acces to to folder and/or file.
This is also the case if the path of the logfile location does not exist.

sbf-, 2018-08-11:
Were you able to fix it?

closed 2018-09-16 by sbf-