SBFspot/SBFspot #233: Fails to compile on Arch Linux ARM

githubcdr, 2018-07-30:
Hi,

Is there anything I can do to resolve this issue?

I'm trying to compile the software on a raspberry pi, running Arch Linux.

```
test -d  || mkdir -p
test -d /../SBFspotUploadCommon || mkdir -p /../SBFspotUploadCommon
test -d /../SBFspot || mkdir -p /../SBFspot
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ ../SBFspotUploadCommon/PVOutput_x.o -s -o sqlite/bin/SBFspotUploadDaemon -Wl,-Bdynamic -lpthread -lcurl
/usr/lib/gcc/armv7l-unknown-linux-gnueabihf/8.1.1/../../../Scrt1.o: In function `_start':
init.c:(.text+0x54): undefined reference to `main'
../SBFspotUploadCommon/PVOutput_x.o: In function `PVOutput::writeCallback(char*, unsigned int, unsigned int, void*)':
PVOutput_x.cpp:(.text+0x2c): undefined reference to `PVOutput::writeCallback_impl(char*, unsigned int, unsigned int)'
collect2: error: ld returned 1 exit status
make: *** [makefile:109: sqlite/bin/SBFspotUploadDaemon] Error 1
```

Thanks for any advice.

sbf-, 2018-07-30:
I'll check it asap. I don't have an idea for the moment.

sbf-, 2018-08-01:
Can you post the installation commands pls?
I don't have the time to look it all up myself.
I know I have to use `pacman` but that's about all
Thanks

sbf-, 2018-08-11:
Sorry, I'm unable to install the needed packages:
```
Welcome to Arch Linux ARM

     Website: http://archlinuxarm.org
       Forum: http://archlinuxarm.org/forum
         IRC: #archlinux-arm on irc.Freenode.net
Last login: Sat Aug 11 20:46:41 2018 from 192.168.1.52
[alarm@alarmpi ~]$ su
Password: 
[root@alarmpi alarm]# pacman -S boost-libs
resolving dependencies...
looking for conflicting packages...

Packages (1) boost-libs-1.62.0-4

Total Download Size:   1.64 MiB
Total Installed Size:  8.35 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
error: failed retrieving file 'boost-libs-1.62.0-4-armv7h.pkg.tar.xz' from mirror.archlinuxarm.org : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded.
```

githubcdr, 2018-08-11:
Sorry for my late response, you probably encountered a cache issue in pacman, best is to perform a full update and then clean the cache as follows

```
pacman -Syy
pacman -Syu
pacman -Scc
```

sbf-, 2018-08-12:
Don't know why, but I already hate Arch Linux :-(
```
[root@alarmpi alarm]# pacman -S boost-libs
resolving dependencies...
looking for conflicting packages...

Packages (1) boost-libs-1.67.0-7

Total Download Size:   1.80 MiB
Total Installed Size:  9.29 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 boost-libs-1.67.0-7-a...  1848.0 KiB  6.68M/s 00:00 [############################] 100%
(1/1) checking keys in keyring                       [############################] 100%
(1/1) checking package integrity                     [############################] 100%
error: boost-libs: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org
>" is unknown trust
:: File /var/cache/pacman/pkg/boost-libs-1.67.0-7-armv7h.pkg.tar.xz is corrupted (invali
d or corrupted package (PGP signature)).
Do you want to delete it? [Y/n] y
error: failed to commit transaction (invalid or corrupted package (PGP signature))
Errors occurred, no packages were upgraded.
```

githubcdr, 2018-08-13:
This quests continues ;)

Please run following to setup keys correctly and try again

```
pacman-key --init
pacman-key --populate archlinux
```

githubcdr, 2018-08-13:
perhaps try https://bbs.archlinux.org/viewtopic.php?id=142893

sbf-, 2018-08-15:
Sorry, but this way I can't support Arch Linux.
I need at least a decent image file and the commands to install the needed packages (bluetooth/boost etc)
For now I used the image I found [here](https://sourceforge.net/projects/archlinux-rpi2/files/)

closed 2018-08-15 by sbf-