SBFspot/SBFspot #235: Forward data via HTTP API?

danbates2, 2018-08-10:
Hi, is is possible to gather the realtime data and forward via http api or MQTT using this? I have an SunnyTriPower 6000TL I'd like to access.

danbates2, 2018-08-10:
My aim is to also post to emoncms.org for visualisation.

sbf-, 2018-08-10:
SBFspot will gather the inverter data into a SQL db, but for uploading it to emoncms.org you will need to write your own uploader. SBFspotUpload daemon only supports PVOutput atm
It's not too difficult when using a python/bash/whatever script together with curl

danbates2, 2018-08-11:
OK thanks, I’ve written a python script to send to emoncms, so could look into this.

danbates2, 2018-08-11:
The core of it is

r = requests.get(url)

where an example url with api key is

https://emoncms.org/input/post?node=emontx&fulljson={"power1":100,"power2":200,"power3":300}

The example below includes reading an apikey from file and incorporates a timer that self corrects to precisely 10s incase of drift. Unnecessary I'm sure for here but I thought to share because I find it useful.

```
import requests
import time

print("Simple code to post data to emonCMS every 10 seconds.")
print("Free to use. OpenEnergyMonitor.org edited by DB, 2018")

# EMONCMS API WRITE KEY IN TEXT FILE OR DELETE THIS CODE AND INPUT DIRECTLY BELOW.
path = "apikey.txt"
with open(path) as f:
        read_data = f.read()
f.closed
apiwritekey = read_data
print("Got API Key, continuing")

# API URLs

# https://emoncms.org/site/api#input
# example: https://emoncms.org/input/post?node=emontx&fulljson={"power1":100,"power2":200,"power3":300}

emoncmsaddress = "emoncms.org"
node_id = "1"
solarpowernow = 111
powernow_string = str (solarpowernow)

# ---single input http api url construction---
emon_api_url_string = "https://" + emoncmsaddress + "/input/post?node=" + node_id + "&fulljson={\"sunnypower\":" + powernow_string + "}" + "&apikey=" + apiwritekey

print("Posting data via: " + emon_api_url_string)

interval = 10  # seconds

def calc_next_time(time_now, interval):
    time_next = (((time_now//interval)*interval)+interval)
    return time_next

time_next = calc_next_time(time.time(), interval)

while(True):

	# Get the current time
	time_now = time.time()

	# Check if it's time to do another transmission
	if time_now >= time_next:

		# Post the data from emonCMS
		r = requests.get(emon_api_url_string)
		#print (r)

		# set the time of next loop
		time_next = calc_next_time(time_now,interval)

	# Don't loop too fast
	time.sleep(0.1)
```

more info here:
https://github.com/emoncms/development/blob/master/Tutorials/Python/PyLink/pylink.py

danbates2, 2018-08-12:
There's also a method of bulk data upload, is that useful here?

danbates2, 2018-08-14:
I've noticed now that this is one of the most popular tools for extracting data from SMA inverters. Congratulations :)
I've tried and failed to get two sma inverter python scripts working from mac osx. I think I'll try SBFSpot on a Pi 3B+..

danbates2, 2018-08-15:
Hello again, I've decided to go all the way and try create a module which interfaces the database created by SBFSpot and sends data to emonCMS locally on a Pi, or remotely to emoncms.org.

emonHub is the project which is the interfacer between various databases/real-time data objects and emonCMS.

Thus I'm liking the idea of creating an emonHub module for SBFSpot, where the SQLite database is read and values sent to emonCMS.

This will be my first (preferably python) programming venture.
Perhaps you can point me to the location of the most recent solar power value (watts) in the SQLite database?

sbf-, 2018-08-15:
> I've noticed now that this is one of the most popular tools for extracting data from SMA inverters. Congratulations :)

Thanks ;-)

> I want to get the SQLite database location of the most recent solar power value (watts), as a starting point
``` SQL
select datetime(TimeStamp, 'unixepoch', 'localtime') TimeStamp,
    Pac1+Pac2+Pac3 PacTot from SpotData
    where Serial=123456789
order by timestamp desc
limit 1;```

danbates2, 2018-08-17:
Here we are!
https://github.com/danbates2/sbfspot_to_emonCMS

danbates2, 2018-08-17:
Needs testing a little longer to be sure. I'll msg again tomorrow. :)

danbates2, 2018-08-18:
All good. https://emoncms.org/graph/embed?graphid=13478

sbf-, 2019-08-04:
@danbates2 fyi there is a dev branch for mqtt

danbates2, 2019-08-10:
@sbf- interesting, thank you.

closed 2020-07-08 by sbf-